innoveit/react-native-ble-manager #609: Properties of a characteristic should have the same structure between iOS/Android

alexsegura, 2020-04-04:
### Version

- react-native-ble-manager v7.2.0
- react-native v0.61.5

---

On iOS, the properties of a characteristic are a zero-indexed array:
```
{
  properties: ["Read", "WriteWithoutResponse", "Write", "Notify", "Indicate"]
}
```

On Android, the properties of a characteristic are an object:
```
{
  properties: {
    "Read":"Read", 
    "WriteWithoutResponse":"WriteWithoutResponse", 
    "Write":"Write", 
    "Notify":"Notify", 
    "Indicate":"Indicate"
  }
}
```

Actually, I don't know if it's really linked to the OS version, but anyway, it would be nice to always return the same data structure. 

IMHO the simplest structure is the zero-indexed array.

If you think it's interesting, I can send a pull-request 🙂 
I would be able to do it only in the JavaScript layer however.

The "problem" is that it would be a BC break, and would need a major version bump I think.

marcosinigaglia, 2020-04-14:
Hi @alexsegura , a PR is welcome.

Loovery, 2022-05-12:
characteristicsMap.putMap("properties", Helper.decodeProperties(characteristic));

Loovery, 2022-05-13:
What is interesting is that only the object with the keys is specified in the types:

`
export interface Characteristic {
    // See https://developer.apple.com/documentation/corebluetooth/cbcharacteristicproperties
    properties: {
      Broadcast?: "Broadcast";
      Read?: "Read";
      WriteWithoutResponse?: "WriteWithoutResponse";
      Write?: "Write";
      Notify?: "Notify";
      Indicate?: "Indicate";
      AuthenticatedSignedWrites?: "AuthenticatedSignedWrites";
      ExtendedProperties?: "ExtendedProperties";
      NotifyEncryptionRequired?: "NotifyEncryptionRequired";
      IndicateEncryptionRequired?: "IndicateEncryptionRequired";
    }
    characteristic: string;
    service: string;
    descriptors?: Descriptor[];
  }
`

closed 2023-12-05 by marcosinigaglia