innoveit/react-native-ble-manager #611: Android .write results in "Write failed"

priyanka-Sadh, 2020-04-15:
### Version

Tell us which versions you are using:

"react": "16.11.0",
 "react-native": "0.62.2",
"react-native-ble-manager": "^7.2.0"
"iOS" : 10+
Android : 10

I am using the code below 
`BleManager.write(id, service, characteristic,[10])
.then(() => {
console.log('Write: ' + data);
})
.catch((error) => {
console.log(error);
});`

I tried timeout,400 and calling retrieveServices before writing.
But not saying reason why it's not working. Error : write failed is not enough to fix it.

marcosinigaglia, 2020-04-15:
Hi, try writeWithoutResponse , if is not working is a device problem, check also che characteristic properties.

priyanka-Sadh, 2020-04-15:
writeWithoutResponse is working on ios and not on android :(
Here are the device characteristics below
`[{"characteristic": "2a00", "properties": {"Read": "Read", "Write": "Write"}, "service": "1800"}, {"characteristic": "2a01", "properties": {"Read": "Read"}, "service": "1800"}, {"characteristic": "2a04", "properties": {"Read": "Read"}, "service": "1800"}, {"characteristic": "6e400003-b5a3-f393-e0a9-e50e24dcca9e", "descriptors": [[Object]], "properties": {"Notify": "Notify"}, "service": "6e400001-b5a3-f393-e0a9-e50e24dcca9e"}, {"characteristic": "6e400002-b5a3-f393-e0a9-e50e24dcca9e", "properties": {"Write": "Write", "WriteWithoutResponse": "WriteWithoutResponse"}, "service": "6e400001-b5a3-f393-e0a9-e50e24dcca9e"}]`

marcosinigaglia, 2020-04-16:
If in iOS works I can't help, try to figure out yourself.

closed 2020-04-16 by marcosinigaglia

Priyankamindcrew, 2020-04-22:
@marcosinigaglia well I had to change library => `react-native-ble-plx` it worked like charm on both ios and android platform.

priyanka-Sadh, 2020-04-22:
@marcosinigaglia well I had to change library => react-native-ble-plx it worked like charm on both ios and android platform.

viren-vii, 2021-06-24:
I had the same problem. I also tried to switching to ble-plx library, but that one didn't work too! Can you help me?