innoveit/react-native-ble-manager #612: Expo can use react-native-ble-manager?

Cupidor, 2020-04-16:
### Version

`[Unhandled promise rejection: TypeError: null is not an object (evaluating 'bleManager.start')]

Stack trace:
  http://127.0.0.1:19001/node_modules%5Cexpo%5CAppEntry.bundle?platform=android&dev=true&minify=false&hot=false:145167:21 in <unknown>
  node_modules\promise\setimmediate\core.js:45:7 in tryCallTwo
  node_modules\promise\setimmediate\core.js:200:23 in doResolve
  node_modules\promise\setimmediate\core.js:66:12 in Promise
  http://127.0.0.1:19001/node_modules%5Cexpo%5CAppEntry.bundle?platform=android&dev=true&minify=false&hot=false:145162:27 in start
  http://127.0.0.1:19001/node_modules%5Cexpo%5CAppEntry.bundle?platform=android&dev=true&minify=false&hot=false:144595:45 in componentDidMount
  node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:17029:10 in commitLifeCycles
  node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:20002:23 in commitLayoutEffects
  node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:307:15 in invokeGuardedCallbackImpl
  node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:531:36 in invokeGuardedCallback
  node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:19768:10 in commitRootImpl
  [native code]:null in commitRootImpl
  node_modules\scheduler\cjs\scheduler.development.js:643:23 in unstable_runWithPriority
  node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:19590:4 in commitRoot
  [native code]:null in commitRoot
  node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:18709:28 in runRootCallback
  [native code]:null in runRootCallback
  node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:5642:32 in runWithPriority$argument_1
  node_modules\scheduler\cjs\scheduler.development.js:643:23 in unstable_runWithPriority
  node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:5638:22 in flushSyncCallbackQueueImpl
  node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:5627:28 in flushSyncCallbackQueue
  node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:18556:30 in scheduleUpdateOnFiber
  node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:11484:17 in dispatchAction
  [native code]:null in dispatchAction
  App.js:22:4 in loadResourcesAndDataAsync
  node_modules\regenerator-runtime\runtime.js:45:44 in tryCatch
  node_modules\regenerator-runtime\runtime.js:274:30 in invoke
  node_modules\regenerator-runtime\runtime.js:45:44 in tryCatch
  node_modules\regenerator-runtime\runtime.js:135:28 in invoke
  node_modules\regenerator-runtime\runtime.js:145:19 in PromiseImpl.resolve.then$argument_0
  node_modules\promise\setimmediate\core.js:37:14 in tryCallOne
  node_modules\promise\setimmediate\core.js:123:25 in setImmediate$argument_0
  node_modules\react-native\Libraries\Core\Timers\JSTimers.js:146:14 in _callTimer
  node_modules\react-native\Libraries\Core\Timers\JSTimers.js:194:17 in _callImmediatesPass
  node_modules\react-native\Libraries\Core\Timers\JSTimers.js:458:30 in callImmediates
  [native code]:null in callImmediates
  node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:407:6 in __callImmediates
  node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:143:6 in __guard$argument_0
  node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:384:10 in __guard
  node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:142:17 in __guard$argument_0
  [native code]:null in flushedQueue
  [native code]:null in invokeCallbackAndReturnFlushedQueue`

Tell us which versions you are using:

- react-native-ble-manager v7.2.0
- react-native v37.0.1

I have see #301  ，you say expo don't support external Native Modules.
So should i do ,expo can use this

marcosinigaglia, 2020-04-16:
I don't get your point. You can't use external native module in Expo.

Cupidor, 2020-04-17:
> I don't get your point. You can't use external native module in Expo.

so expo can not use react-native-ble-manager?

Cupidor, 2020-04-17:
i catch Error writing status: 13 ，what is it mean

closed 2020-04-20 by Cupidor

viren-vii, 2021-04-12:
Hello! I am new to react native. I used expo and got the same error. What is another way to use react-native-ble-manager? Please guide me :)

lnCrew, 2021-06-09:
I'm also new with expo and I have the same question, is there an alternative to react-native-ble-manager if I can't use it with expo ?

Cupidor, 2021-06-10:
> I'm also new with expo and I have the same question, is there an alternative to react-native-ble-manager if I can't use it with expo ?

finally i use react native, expo maybe cant react-native-ble-manager

Cupidor, 2021-06-10:
> react native

react native can use ,expo can not  use toothblue module