innoveit/react-native-ble-manager #616: Connect to device by getBondedPeripherals() function

erikthiago, 2020-04-21:
### Version

Tell us which versions you are using:

- react-native-ble-manager v7.2.0
- react-native v0.61.4
- Android v9

### Expected behaviour
Connect to device by bonded list

Tell us what should happen

Generic "Connection Error"

### Steps to reproduce

1. Get bonded devices list
2. Select one
3. Try connect

### Code

 let Bonded = () => {

        BleManager.start({ showAlert: false });

        BleManager.getBondedPeripherals()
            .then((bondedPeripheralsArray: any) => {
                setLista(bondedPeripheralsArray);
            }).catch((error) => {
                console.log(error.message);
            });
    }

    let Connect = async (item: any) => {

        BleManager.checkState();

        BleManager.connect(item.item.id)
            .then(() => {
                console.log("chegou aqui connect");

                // Success code
                ToastAndroid.showWithGravity(
                    "Conectado ao dispositivo!",
                    ToastAndroid.SHORT,
                    ToastAndroid.CENTER
                );
            })
            .catch((error) => {
                // Failure code
                ToastAndroid.showWithGravity(
                    "Erro: " + error,
                    ToastAndroid.SHORT,
                    ToastAndroid.CENTER
                );

                console.log(error);
            });
    }

 let RenderSeparator = () => {
        return (
            <View
                style={{
                    height: 1,
                    width: "86%",
                    backgroundColor: "#CED0CE",
                    marginLeft: "14%"
                }}
            />
        );
    };

    let RenderItem = (item: any) => {
        const color = item.connected ? 'green' : '#fff';

        return (
            <TouchableHighlight onPress={() => Connect(item)} style = {styles.touch}>
                <View style={[styles.row, { backgroundColor: color }]} >
                    <Text style={{ fontSize: 12, textAlign: 'center', color: '#333333', padding: 10 }}>{item.item.name}</Text>
                    <Text style={{ fontSize: 12, textAlign: 'center', color: '#333333', padding: 10 }}>{item.item.id}</Text>
                </View>
            </TouchableHighlight>
        );
    }

    return (
        <View style={styles.container}>
            <TouchableOpacity onPress={Bonded} style={styles.button}>
                <Text style={styles.buttonText}>Dispositivos</Text>
            </TouchableOpacity>

            <FlatList 
                data={lista}
                renderItem={({ item }) => <RenderItem item={item} />}
                keyExtractor={(item, index) => index.toString()}
                ListEmptyComponent={() =>
                    <Text style={styles.buttonText}>
                        Não há itens!
                    </Text>
                }
                ItemSeparatorComponent={RenderSeparator}
            />
        </View>
    );
}

erikthiago, 2020-04-21:
This operations is supported by this lib?

marcosinigaglia, 2020-05-07:
Hi, I don't understand your question.

erikthiago, 2020-05-10:
I'm getting a list of bonded devices. I want to choose one and connect. But this operation return generic connection error. So, i want to know if this operation is supported in react-native-ble-manager.  I'm using the above code.

gbalduzzi, 2021-03-10:
Hi @marcosinigaglia , thank you very much for your job maintaining this package.

I'm having the same issue described by @erikthiago .

Basically, in my app I want to connect on bluetooth devices that are already paired with the android smartphone, in order to avoid scanning and thus requesting the user location. By calling `BleManager.getBondedPeripherals()` I correctly obtain the list of paired devices. When I try to connect to them though, I don't receive any answer.

```
try {
      let connectionResult = await BleManager.connect(device.id);
      console.log(connectionResult);
    } catch (err) {
      console.warn(err);
    }
```

I get no log and no warning, even waiting for a long amount of time. 

Am I required to scan for devices to connect to a paired device?

Thanks

marcosinigaglia, 2021-03-22:
Hi @gbalduzzi , I never worked with bonded device so I'm not sure that the code is perfect. The list of the peripherals is stored [here](https://github.com/innoveit/react-native-ble-manager/blob/e7d83a503fd351f7a84275748391bffd05e11a55/android/src/main/java/it/innove/BleManager.java#L77) but the getBondedPeripherals function is not updating the list so to use the peripheral you need to scan before. Try to change the method adding the missing peripheral in the list.

closed 2023-12-06 by marcosinigaglia