innoveit/react-native-ble-manager #618: searching only one device

syedjawadhashmi, 2020-05-05:
Hi, 
I have integrated  this libraray successfully and done all its configuration but its searching only one device and its not connecting at all , please let me know the issues i am facing

marcosinigaglia, 2020-05-07:
How many device are in the BLE range? Do you have an error during the connection, what is the device are you trying to connect?

syedjawadhashmi, 2020-05-14:
only 1 device is scan which is my pc bluetooth other bluetooth devices are not scan in the app but in mobile bluetooth setting showing all the devices and connect successful but in the app using this module bluetooth scanning and connection is not working

viratbhatt, 2020-12-29:
For me it shows bunch of random Bluetooth devices but my Other devices are not visible in the scan

kkureli, 2021-11-22:
> For me it shows bunch of random Bluetooth devices but my Other devices are not visible in the scan
same

closed 2023-12-06 by marcosinigaglia