innoveit/react-native-ble-manager #619: have an issue with finding the right characteristics of a ble device. needs urgent help

dotkom2, 2020-05-07:
### Version

Tell us which versions you are using:

- react-native-ble-manager v?.?.?
- react-native v?.?
- iOS/Android v.?

### Expected behaviour

Tell us what should happen

### Actual behaviour

Tell us what happens instead

### Steps to reproduce

1.
2.
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

marcosinigaglia, 2020-05-14:
Hi, can you explain better?

dotkom2, 2020-05-15:
Hi, Yes. I'm facing the problem of not able to read the weight value from
the BLE device. It gives null for some characteristic and for some
characteristic gives ascii numbers that says serial number or manufacturer
name. So how do I know which characteristic will give value and how? Your
help will be much appreciated.

marcosinigaglia, 2020-05-15:
You need to check the BLE device documentation.

closed 2020-05-15 by marcosinigaglia