innoveit/react-native-ble-manager #622: Data not received by Android of lower version(5.1.0)

qingjiaowochengzd, 2020-05-13:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.8
- react-native v0.55.4
- iOS/Android v.5.1.0  4.4.4

### Expected behaviour

IOS and android v8.0.0 can receive no problem.

android v5.1.0  after   startNotification 
can not listen data 

bleManagerEmitter.addListener(
        'BleManagerDidUpdateValueForCharacteristic',
        ({ value, peripheral, characNotif, service }) => {
            console.log("received a notif", value)
        }
);

### Actual behaviour

IOS and android v8.0.0 can get data

### Steps to reproduce

connectBle = async (peripheral) => {
    try {
      await BleManager.connect(peripheral.id)
      console.log('连接成功', peripheral)

      const peripheralInfo = await BleManager.retrieveServices(peripheral.id)
      console.log(peripheralInfo)

      let serviceUuid = '49535343-fe7d-4ae5-8fa9-9fafd205e455'
      let characteristicUuid = "49535343-1e4d-4bd9-ba61-23c647249616"

      const Notification = await BleManager.startNotification(peripheral.id, serviceUuid, characteristicUuid)

      bleManagerEmitter.addListener(
        'BleManagerDidUpdateValueForCharacteristic',
        ({ value, peripheral, characNotif, service }) => {
            console.log("received a notif", value)
        }
      );

    } catch (err) {
      console.log(err)
      Toast.fail(err, 2, null, false)
    }

  }

### Stack trace and console log

```
```

marcosinigaglia, 2020-05-14:
Hi, we tested on android 4.4.2 so probably is your device that have son problem.

llMrRongll, 2020-07-28:
> ### Version
> Tell us which versions you are using:
>
> * react-native-ble-manager v6.6.8
> * react-native v0.55.4
> * iOS/Android v.5.1.0  4.4.4
>
> ### Expected behaviour
> IOS and android v8.0.0 can receive no [...]

I met the same problem

reworldgit, 2020-10-09:
same problem
In Android 10 works well
In Android 5.1
calling Blemanager.connect() no return. no success no fail.

closed 2023-12-04 by marcosinigaglia