innoveit/react-native-ble-manager #625: Android 10 Bluetooth cannot be searched ?

Lizhooh, 2020-05-20:
### Version

Tell us which versions you are using:

- react-native-ble-manager latest
- react-native v0.60.2
- iOS/Android v10

marcosinigaglia, 2020-05-21:
Hi, you have to debug in your device.

qsayeed, 2020-07-26:
For Android 10, you need to give both permission  **ACCESS_FINE_LOCATION** and **ACCESS_COARSE_LOCATION**.

sasurau4, 2020-08-06:
FIY: This link is helpful for users met this issue.
https://developer.android.com/guide/topics/connectivity/bluetooth#Permissions

anwarhamr, 2020-08-12:
Possible related https://github.com/innoveit/react-native-ble-manager/issues/647 I have made the recommended change listed here but didn't help me.

Theofilos-Chamalis, 2020-08-13:
I had the same issue and adding `<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />` in the AndroidManifest.xml file as @qsayeed mentioned has solved the issue for me.

anwarhamr, 2020-08-17:
> I had the same issue and adding `<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />` in the AndroidManifest.xml file as @qsayeed mentioned has solved the issue for me.

That change did not work for me.  Still not getting scan results.

kapernikxd, 2020-10-21:
i have  this code in AndroidManifest.xml 
'
	<uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
'

But it still doesn't work in Android 10 qp1a.

kapernikxd, 2020-10-21:
it will be solution for me https://github.com/Polidea/react-native-ble-plx/issues/730

WookieFPV, 2020-10-23:
you have to request the permission "android.permission.ACCESS_FINE_LOCATION" and have it in the AndroidManifest.xml file.
This fixed the issue for me

WookieFPV, 2020-11-01:
@kapernikxd do you check and request the ACCESS_FINE_LOCATION permission at runtime?
You need this for Android 10.
(You don't have to request ACCESS_COARSE_LOCATION if you have ACCESS_FINE_LOCATION)

closed 2020-11-18 by marcosinigaglia