innoveit/react-native-ble-manager #628: Characteristics empty in response of retrieveServices call.

vasanthonline, 2020-05-27:
### Version

Tell us which versions you are using:

- react-native-ble-manager v7.2.0
- react-native v0.62.2
- iOS/Android v.Android v10

### Expected behaviour

Am using a WEP MTP-25 printer - https://www.wepmyshop.com/wep-mtp-25.
The objective is to print receipts via this device.

Am able to make a successful connection to the device. After the connection is established call is made to `retrieveServices` to get the peripheralInfo.
Below is the response of the peripheralInfo - 
`{"advertising": {"isConnectable": true, "localName": "MTP-25", "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "AgEKAgoEAwLwGBEG8sPwrqn6FYydSa5zcQqB5wcJTVRQLTI1AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {}, "serviceUUIDs": ["18f0", "e7810a71-73ae-499d-8c15-faa9aef0c3f2"], "txPowerLevel": 4}, "characteristics": [], "id": "DC:0D:30:2F:F2:63", "name": "MTP-25", "rssi": -33, "services": []}`

Expect a charecteristic ID / UUID to use in `startNotification`.

### Question?
The `characteristics` array is empty.  What should be the `characteristicUUID` passed to`sendNotification` call?
 

### Steps to reproduce
Am using the [example](https://github.com/innoveit/react-native-ble-manager/tree/master/example) code in the repo, as it is.

### Stack trace and console log

[image]

marcosinigaglia, 2020-05-28:
Check the device documentation.

tarkcelk, 2021-07-13:
Hi , people are writing here because the documentation couldn't make sense , please help us out here , how we can get serviceUUID and characteristicUUID of connected peripheral , in documentation they're static