innoveit/react-native-ble-manager #629: No way to increase write() max byte size? (trying to implement OTA Update for device)

NeilMcGuiness, 2020-05-28:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.7.0
- react-native v61.4

### Problem

I suppose this is a feature request of a kind, but max write size of 20 seems very small doesn't it?

I am trying to develop Over The Air update functionality for my app whereby i can stream a hex file (~ 700 kb) to my device in order to update the firmware, however this will take a long time being limited to 20 bytes at a time.

Is there no way to increase this 20 byte limit? I'm fairly sure the cypress app (compatible with the Peripheral device we're using) sends larger chunks of data much quicker to do OTA updating.

If it's possible to adjust this in the source code, just a point in the right direction would be very much appreciated.

Thanks in advanced.

marcosinigaglia, 2020-05-29:
Hi, use [requestMTU](https://github.com/innoveit/react-native-ble-manager#requestmtuperipheralid-mtu-android-only-api-21)

NeilMcGuiness, 2020-05-29:
Yes I found this today, my mistake, apologies! I'll close this now.

closed 2020-05-29 by NeilMcGuiness