innoveit/react-native-ble-manager #630: name is always null

fazeelhaider, 2020-05-31:
### Version

Tell us which versions you are using:

- react-native-ble-manager v^7.2.0
- react-native v0.62.2
- iOS/Android v.9

I'm using this example: https://github.com/innoveit/react-native-ble-manager/blob/master/example/App.js

### Expected behaviour
show nearby peripherals

### Actual behaviour
name of peripherals always null and not showing actual nearby peripherals

 {"advertising": {"isConnectable": true, "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "AgEGE/9MAAwOCKmRF0YL7OjksF2yGnkAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {}, "serviceUUIDs": [], "txPowerLevel": -2147483648}, "id": "47:8D:BD:3C:82:35", "name": null, "rssi": -68}

Tell us what happens instead

### Steps to reproduce

1. npm i --save react-native-ble-manager

2.  <uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />

3. https://github.com/innoveit/react-native-ble-manager/blob/master/example/App.js

marcosinigaglia, 2020-06-04:
The device name is set on the device.

closed 2020-06-04 by marcosinigaglia

gragsie, 2020-06-26:
Yes this is an bug. The name is always returned on iOS, but on Android it is *mostly* null for the exact same peripherals.