innoveit/react-native-ble-manager #632: This library is not working. No response. Only UI is showing.

SumeriLal, 2020-06-04:
### Version

Tell us which versions you are using:

- react-native-ble-manager v7.2.0
- react-native v0.62.2
- Android v10 and Android v9 pie

### Expected behaviour

It should scan all nearby available devices.

Tell us what should happen: It should show all available bluetooth active devices nearby the phone.

### Actual behaviour
It showing 'No perioherals' after clicking on can button and console log showing 'Stop Scanning'. 

Tell us what happens instead:

Console log showing 'Scanning...' after one second it showing 'Scan is stopped'. But in my phone screen there is "No peripherals".

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

[image]
[image]

Please Help me!

Thank you!

marcosinigaglia, 2020-06-04:
Hi, this is not an issue.

closed 2020-06-04 by marcosinigaglia

zhubinsheng, 2020-09-01:
I'm really sorry to disturb your time. I also have a similar problem. The log is as follows:

D/ReactNativeBleManager: state:on

I/ReactNativeJS: 'BleManagerDidUpdateStatea:', { state: 'on' }

D/ReactNativeBleManager: scan

D/ReactNativeBleManager: scan start

scan start2

scan start3

D/BluetoothAdapter: isLeEnabled(): ON

D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=5 mScannerId=0

D/ReactNativeBleManager: scan start34

I/ReactNativeJS: Scan started

D/ReactNativeBleManager: scan start4

D/BluetoothAdapter: isLeEnabled(): ON

I/ReactNativeJS: 'BleManagerStopScan:', 'Scanning is stopped'

My references in the original Android project will appear

zhubinsheng, 2020-09-01:
Sorry, I lost the following permissions to add. After I added, the mixed development application runs perfectly

<uses-permission android:name= " android.permission.INTERNET " />

<uses-permission android:name= " android.permission.BLUETOOTH " />

<uses-permission android:name= " android.permission.BLUETOOTH_ ADMIN" />

<uses-permission android:name= " android.permission.ACCESS_ COARSE_ LOCATION" />

<uses-permission android:name= " android.permission.ACCESS_ FINE_ LOCATION" />

venkatesh-kyyba, 2020-09-16:
I am also facing the same issue,
I am not able to get the peripheral from Android.
can you please let me know what permissions you added.
please let me know how to resolve the issue.

Datagrdn, 2020-10-07:
@zhubinsheng What are the permissions you added?

I am having the same issue. When I press scan I can see on my peripheral that it is trying to connect but then the scan stops.

Anyone else have a solution to this?

Datagrdn, 2020-10-07:
Ah I see you were referring to the permissions in the ReadMe. I added those and was still having the same issue.

I fixed it by adding the code in the second answer here:

https://stackoverflow.com/questions/32708374/bluetooth-le-scan-fails-in-the-background-permissions