innoveit/react-native-ble-manager #633: requestMTU sucess, but BleManagerDidUpdateValueForCharacteristic still receieved 20 characters

preflower, 2020-06-04:
### Version

Tell us which versions you are using:

- react-native-ble-manager v7.2.0
- react-native v0.62.2
- Android v9

### Expected behaviour

BleManagerDidUpdateValueForCharacteristic receives more than 20 characters.

### Actual behaviour

BleManagerDidUpdateValueForCharacteristic receives max 20 characters, and subcontracted my data

### Stack trace and console log

[Thu Jun 04 2020 18:23:54.481]  LOG      Scan started
[Thu Jun 04 2020 18:23:56.900]  LOG      MTU size changed to 128 bytes
[Thu Jun 04 2020 18:23:59.920]  LOG      Write success:  [58, 163, 0, 0, 0, 0, 0, 1, 0, 2, 128, 1, 131, 168]
[Thu Jun 04 2020 18:23:59.371]  LOG      Recieved :£AW3JVP for characteristic 0000ffe1-0000-1000-8000-00805f9b34fb
[Thu Jun 04 2020 18:23:59.378]  LOG      Recieved 0A21500082	Ò for characteristic 0000ffe1-0000-1000-8000-00805f9b34fb

```
```

marcosinigaglia, 2020-06-04:
Probably is a device problem.

preflower, 2020-06-05:
> Probably is a device problem.

i try to merge package manually, but how to get the package's header?

closed 2020-09-17 by preflower

15638228357, 2021-05-26:
I also encountered the same problem. I successfully modified the Mtu but still sent up to 20 bytes. How did you solve it?