innoveit/react-native-ble-manager #638: Peripheral.java in Release 7.3.0 looks wrong.

Loxrie, 2020-06-12:
### Version

Tell us which versions you are using:

- react-native-ble-manager v7.3.0
- react-native v0.61.4
- Android Any

### Expected behaviour

My disconnect handler should be called.

### Actual behaviour

The disconnect handler is not called, nor is bonding.

### Steps to reproduce

Use 7.3.0, fail to connect to a device.  For example by cancelling the pairing request.

### Stack trace and console log

None.

It seems to me that the code here:

```
                gatt = gatta;

		if (status != BluetoothGatt.GATT_SUCCESS) {
		    // TODO: cancel the service discovery runnable
		    gatt.close();

		} else // if ...
```

https://github.com/innoveit/react-native-ble-manager/blob/b7c26343b97d12da3d3aa7f6e64dad520d6f1ae9/android/src/main/java/it/innove/Peripheral.java

And released in 7.3.0, is wrong.  The comment there after else implies the code is not complete.  What happens instead is that it calls gatt.close() then skips the if statements afterwards such as:

```
if (newState === VluetoothProfile.STATE_CONNECTED)
```
and
```
} else if (newState == BluetoothProfile.STATE_DISCONNECTED) {
```

Loxrie, 2020-06-12:
This code is also throwing an exception:

```
if (android.os.Build.VERSION.SDK_INT >= Build.VERSION_CODES.O){
				// We can check if peripheral is connectable using the scanresult
				advertising.putBoolean("isConnectable", scanResult.isConnectable());
			} else{
				// We can't check if peripheral is connectable
				advertising.putBoolean("isConnectable", true);
			}
```

```
java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.bluetooth.le.ScanResult.isConnectable()' on a null object reference
    at it.innove.LollipopPeripheral.asWritableMap(LollipopPeripheral.java:44)
    at it.innove.BleManager$1.onReceive(BleManager.java:542)
    at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$-android_app_LoadedApk$ReceiverDispatcher$Args_52226(LoadedApk.java:1319)
    at android.app.-$Lambda$FilBqgnXJrN9Mgyks1XHeAxzSTk.$m$0(Unknown Source:4)
    at android.app.-$Lambda$FilBqgnXJrN9Mgyks1XHeAxzSTk.run(Unknown Source:0)
    at android.os.Handler.handleCallback(Handler.java:789)
    at android.os.Handler.dispatchMessage(Handler.java:98)
    at android.os.Looper.loop(Looper.java:164)
    at android.app.ActivityThread.main(ActivityThread.java:6944)
    at java.lang.reflect.Method.invoke(Native Method)
    at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:327)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1374)
```

andrewcharnley, 2020-06-14:
I'm having the same issue with disconnect now being called.

marcosinigaglia, 2020-06-17:
Hi, a PR is welcome, sorry I don't have tested the other PR

Loxrie, 2020-06-24:
I'll hopefully be able to do one soon, should get a weekend free soon.

glenne, 2020-06-28:
I have the same issue here.  The fix is to always close gatt if status indicates an error and always process the actions for connect and disconnect so notifications are published.  I've tested this quite a bit with various disconnect/connect conditions and it works well.

I'll post a pull request shortly.

commit 11020276 (2020-06-29): Merge pull request #644 from glenne/master

Fix #638 - Restore calling of disconnect handlers
commit a26b2d55 (2020-06-29): Fix #638 - Restore calling of disconnect handlers
closed 2020-06-29 by marcosinigaglia