innoveit/react-native-ble-manager #645: 'BleManagerDisconnectPeripheral' listener not being triggered on Android

cklopotic, 2020-06-29:
### Version

Tell us which versions you are using:

- react-native-ble-manager v7.3.0
- react-native v0.62.2
- iOS/Android: Android (v10), iOS (13.5.1)

### Expected behavior
When a connected Peripheral goes out of range/powered off, the specified callback function for the "BleManagerDisconnectPeripheral" listener should be called.
bleManagerEmitter.addListener('BleManagerDisconnectPeripheral', handleDisconnectedPeripheral )

### Actual behavior

On iOS, this works as expected.  On Android the callback is never triggered.  This used to work on an earlier version of my app, since then I've updated all my libraries and haven't been able to get it working again.  I've been troubleshooting this for weeks and narrowed it down to this specific behavior.  I had also ran my app in Android Studio and debugged in there.  Pulling open LogCat I can see that my device recognizes the BLE peripheral has been disconnected, but that information isn't making its way into this library.
I very well could be doing something wrong, or interfacing incorrectly between iOS & Android.

### Steps to reproduce

1.  Create a new react-native init project
2. Install the react-native-ble-manager library into your newly created project, & Pod Install
3. Edit your Android Manifest & Info.plist file for appropriate bluetooth permissions
4. Replace your App.js file in your project with the App.js file located in the [react-native-ble-manager/example](https://github.com/innoveit/react-native-ble-manager/blob/master/example/App.js)
    -- example App.js a component class, my app is a functional component with hooks.  Both situations demonstate the same behavior.
5. Run on an actual device.
  - Scan for devices
  - pick a known peripheral you can connect to & connect (connecting and manual disconnecting work fine)
  - while connected to your known peripheral in the example app (your peripheral should be green), power the peripheral down.
     -- iOS - captures the event that the peripheral has disconnected, and disconnects the peripheral in the app (no longer green)
    -- Android - logCat shows that the peripheral has disconnected, the sample app does not.  No error is thrown.  Peripheral in the app stays green signaling that it is still connected.

cklopotic, 2020-06-29:
After I wrote this I did some more troubleshooting with the example.
If I run react-native-ble-manager v 7.2.0 it works as expected.  Killing the metro sever, install v 7.3.0, restart app, problem comes back.  Downgrade again to v 7.2.0 and it works as expected again.

Suspecting something changed from v7.2.0 to v7.3.0

smartls, 2020-06-30:
I meets same issue in android, first connect a device with my app, then remove a paired device from mobile, the ble device was disconnected at once, but BleManagerDisconnectPeripheral was not triggered. 
So I checked the source code, find the issue:
`
@Override
public void onConnectionStateChange(BluetoothGatt gatta, int status, int newState) {

	Log.d(BleManager.LOG_TAG, "onConnectionStateChange to " + newState + " on peripheral: " + device.getAddress()
			+ " with status " + status);

	gatt = gatta;

	if (status != BluetoothGatt.GATT_SUCCESS) {
		// TODO: cancel the service discovery runnable
		gatt.close();

	} else // if ...

	if (newState == BluetoothProfile.STATE_CONNECTED) {
		connected = true;

		new Handler(Looper.getMainLooper()).post(new Runnable() {
			@Override
			public void run() {
				try {
					gatt.discoverServices();
				} catch (NullPointerException e) {
					Log.d(BleManager.LOG_TAG, "onConnectionStateChange connected but gatt of Run method was null");
				}
			}
		});

		sendConnectionEvent(device, "BleManagerConnectPeripheral", status);

		if (connectCallback != null) {
			Log.d(BleManager.LOG_TAG, "Connected to: " + device.getAddress());
			connectCallback.invoke();
			connectCallback = null;
		}

	} else if (newState == BluetoothProfile.STATE_DISCONNECTED) {

		this.disconnect(true);

		sendConnectionEvent(device, "BleManagerDisconnectPeripheral", status);
		List<Callback> callbacks = Arrays.asList(writeCallback, retrieveServicesCallback, readRSSICallback,
				readCallback, registerNotifyCallback, requestMTUCallback);
		for (Callback currentCallback : callbacks) {
			if (currentCallback != null) {
				currentCallback.invoke("Device disconnected");
			}
		}
		if (connectCallback != null) {
			connectCallback.invoke("Connection error");
			connectCallback = null;
		}
		writeCallback = null;
		writeQueue.clear();
		readCallback = null;
		retrieveServicesCallback = null;
		readRSSICallback = null;
		registerNotifyCallback = null;
		requestMTUCallback = null;
	}
}
`
the issue was happen in the code "if (status != BluetoothGatt.GATT_SUCCESS)" condition, in my situation, state is 22, code never run into the STATE_DISCONNECTED condition, only in this condition, BleManagerDisconnectPeripheral was triggered.

smartls, 2020-06-30:
It's a bug in version 7.3.0, I saw a new PR was mergered into Peripheral.java, but not published on NPM, suggest to use other lib, or revert to version 7.2.0

closed 2020-06-30 by marcosinigaglia