innoveit/react-native-ble-manager #647: Scanning and getDiscoveredPeripherals not working

Balajisivakumar92, 2020-07-16:
7.3.1 Version

Tell us which versions you are using:

- react-native-ble-manager 7.3.1
- react-native "0.62.2"
- Android 10

###Expected behaviour

Scanning and getDiscoveredPeripherals, have to give available nearby devices. It was working well in debug mode.

###Actual behaviour

Scanning and getDiscoveredPeripherals, It not working now, After Generating the release APK and now its not working for debug mode also.

kiafaldorius, 2020-07-22:
Restart your phone to see if it's working again. I've seen cases where Android's bluetooth subsystem just randomly stops servicing or returning data.

qsayeed, 2020-07-25:
Its working on android 7 and 8. But on android 10, its not discovering peripherals.

anwarhamr, 2020-08-12:
I too am seeing this issue.
Had version of ble 7.1.3 was working yesterday 
Today, upgraded to android 10 on tablet that the app was running on and working just fine for months.
Restarted app, restarted tablet, reset network settings (reset bluetooth settings)
Still nothing.
Upgraded to ble 7.3.1 rebooted, reinstalled and still not working
Tablet is Samsung Galaxy Tab S5e

Debugging: 
I see that ble starts.
I get response from scan but devices are never discovered. 
I have another tablet that is android 6 which will be our production device.  Same code works just fine.  
What else can I try?

kiafaldorius, 2020-08-16:
Try:

```
BleManager.scan([], 15, false, {
  numberOfMatches: 3,
  matchMode: 1,
  scanMode: 2,
  reportDelay: 0
})
```

anwarhamr, 2020-08-17:
> Try:
> 
> ```
> BleManager.scan([], 15, false, {
>   numberOfMatches: 3,
>   matchMode: 1,
>   scanMode: 2,
>   reportDelay: 0
> })
> ```

Didn't help, thanks for the idea.

anwarhamr, 2020-08-18:
I've tried to capture the scan via Logcat here is the results.  

08-17 17:57:54.963 25441 25512 D ReactNativeBleManager: scan
08-17 17:57:54.965 25441 25512 D ReactNativeBleManager: Filter service: 48BBE7D5-81E5-46A4-B3AD-23417934D96D
08-17 17:57:54.965 25441 25512 D BluetoothAdapter: STATE_ON
08-17 17:57:54.966 25441 25512 I chatty  : uid=10262(com.epitel.remi_tablet) mqt_native_modu identical 1 line
08-17 17:57:54.966 25441 25512 D BluetoothAdapter: STATE_ON
08-17 17:57:54.968 25441 25512 D BluetoothAdapter: BLE support array set: 010011
08-17 17:57:54.968 25441 25512 D BluetoothLeScanner: Start Scan with callback
08-17 17:57:54.969 16868 16950 D BtGatt.ContextMap: add() - appUid: 10262, appPid: 25441, appName: com.epitel.remi_tablet
08-17 17:57:54.969 16868 16957 I bt_stack: [INFO:gatt_api.cc(953)] GATT_Register c9b0602a-d9ac-433e-8ade-795c9f247c03
08-17 17:57:54.970 16868 16957 I bt_stack: [INFO:gatt_api.cc(990)] allocated gatt_if=9
08-17 17:57:54.971 25441 25460 D BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=9 mScannerId=0
08-17 17:57:54.971 16868 16950 D BluetoothAdapter: STATE_ON
08-17 17:57:54.972 16868 16950 E BluetoothUtils: Permission denial: Location is off.
08-17 17:57:54.975 16868 16915 D BluetoothAdapter: STATE_ON
08-17 17:57:54.978 25441 25511 I ReactNativeJS: '%c Dispatching ', 'background: #222; color: #bada55', { type: 'BLE_SCAN_STARTED',
08-17 17:57:54.978 25441 25511 I ReactNativeJS:   bondedPeripherals: [],
08-17 17:57:54.978 25441 25511 I ReactNativeJS:   discoveredPeripherals: [],
08-17 17:57:54.978 25441 25511 I ReactNativeJS:   utc: 1597708674977 }
08-17 17:57:54.980 16868 16916 E BtGatt.GattService: [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_START, appName: com.epitel.remi_tablet, scannerId: 9, reportDelayMillis=0
08-17 17:57:54.980 16868 16915 D BluetoothAdapter: STATE_ON
08-17 17:57:54.981 16868 16915 D BluetoothAdapter: STATE_ON
08-17 17:57:54.982   695   695 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK ON
08-17 17:57:54.984   695   695 D vendor.qti.bluetooth@1.0-wake_lock: Acquire wakelock is acquired 
08-17 17:57:54.984   695   695 I vendor.qti.bluetooth@1.0-ibs_handler: DeviceWakeUp: Writing IBS_WAKE_IND
08-17 17:57:54.987   695 16955 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_ACK: 0xFC
08-17 17:57:54.988   695 16955 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
08-17 17:57:54.988   695 16955 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
08-17 17:57:54.990 16868 16915 D BtGatt.ScanManager: configureFilterParamter() : onFoundTimeout | onLostTimeout | onFoundCount | numOfTrackingEntries | scanFilterRssiThreshold | trackableScanFilterRssiThreshold | matchMode
08-17 17:57:54.990 16868 16915 D BtGatt.ScanManager: configureFilterParamter() : __________500__|________10000__|___________1__|___________________0__|___________________-128__|____________________________-128__|________1_
08-17 17:57:54.993 16868 16915 D BtGatt.ScanManager: configureRegularScanParams() scan param set to 6553 / 6553 (units)
08-17 17:57:54.993 16868 16957 W bt_btm  : btm_ble_disable_resolving_list() rl_state = 0x0, rl_mask = 0x2, to_resume = 1
08-17 17:57:54.993 16868 16957 W bt_btm  : btm_ble_disable_resolving_list() rl_state = 0x0, rl_mask = 0x2, to_resume = 1
08-17 17:57:55.003 25441 25511 I ReactNativeJS: '%c Action with type "BLE_SCAN_STARTED" took 24.00 milliseconds.', 'background: #bada55; color: #222'
08-17 17:57:55.012 25441 25511 I ReactNativeJS: '%c Dispatching ', 'background: #222; color: #bada55', { type: 'BLE_SCAN_REQ_CMPLT',
08-17 17:57:55.012 25441 25511 I ReactNativeJS:   value: 'BLE_STATE_UPDATED',
08-17 17:57:55.012 25441 25511 I ReactNativeJS:   location: 'BLE_STATE_UPDATED',
08-17 17:57:55.012 25441 25511 I ReactNativeJS:   scanningForServices: [ '48BBE7D5-81E5-46A4-B3AD-23417934D96D' ],
08-17 17:57:55.012 25441 25511 I ReactNativeJS:   utc: 1597708675011 }
08-17 17:57:55.012 25441 25511 I ReactNativeJS: '%c Action with type "BLE_SCAN_REQ_CMPLT" took 0.00 milliseconds.', 'background: #bada55; color: #222'
08-17 17:57:55.037   695 16955 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
08-17 17:57:55.316  1339 25541 D WifiConnectivityMonitor.DnsPingerHandler: setCurrentLinkProperties: lp={InterfaceName: wlan0 LinkAddresses: [ fe80::d845:c5ff:feee:cf5a/64,192.168.0.197/16,2605:a601:a982:c400:d845:c5ff:feee:cf5a/64,2605:a601:a982:c400:856f:9754:b7ae:2c61/64 ] DnsAddresses: [ /fe80::f6f5:e8ff:fe6e:dd82%wlan0,/192.168.1.1 ] Domains: null MTU: 0 TcpBufferSizes: 524288,1048576,4194304,524288,1048576,4194304 Routes: [ fe80::/64 -> :: wlan0,::/0 -> fe80::f6f5:e8ff:fe6e:dd82 wlan0,2605:a601:a982:c400::/64 -> :: wlan0,192.168.0.0/16 -> 0.0.0.0 wlan0,0.0.0.0/0 -> 192.168.1.1 wlan0 ]}
08-17 17:57:55.630  1339 25542 D WifiConnectivityMonitor.DnsThread:  [|210] []
08-17 17:57:55.632  1339  1717 D WifiConnectivityMonitor: DNS resultType : 0, responseTime : 210
08-17 17:57:55.633  1339  1716 D WifiConnectivityMonitor: updateBssidLatestDnsResultType - result: 0
08-17 17:57:55.807   695 16955 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
08-17 17:57:55.807   695 16955 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
08-17 17:57:55.815 16868 16901 D BluetoothUtils: getBtEnabledContainers(): btContainers = []
[... cut ...]
08-17 17:57:55.930   504   504 I hwservicemanager: getTransport: Cannot find entry vendor.samsung.hardware.radio.channel@2.0::ISehChannel/imsd in either framework or device manifest.
08-17 17:57:55.997   695 16960 I vendor.qti.bluetooth@1.0-ibs_handler: DeviceSleep: TX Awake, Sending SLEEP_IND
08-17 17:57:55.997   695 16960 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK OFF
08-17 17:57:56.147   695 16928 D vendor.qti.bluetooth@1.0-wake_lock: Release wakelock is released

kiafaldorius, 2020-08-25:
> 08-17 17:57:54.972 16868 16950 E BluetoothUtils: Permission denial: Location is off.

Well, there's your problem. Location services are off on that phone or you don't have permission.

anwarhamr, 2020-08-25:
Something is changing it then, this works i just run out of clients at some point.?.?

kiafaldorius, 2020-08-28:
> run out of clients at some point

I don't know what you mean by that. Basic process is to double check the AndroidManifest.xml, here are the Bluetooth permissions I have in mine on my react-native 0.61.5 app:

```
    <uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
```

On one of my apps on react-native 0.62.2, I don't have any bluetooth uses-permission entries and it appears to still work...

Then check for location permissions in the settings:

Settings -> Location

Use location switcher at the top should be enabled.

Then tap App Permission, make sure your app is in one of the first two lists. (Allowed all the time or Allowed only while in use)

PS. Your scan also seems to be scanning for a particular service, did you run my code above exactly as is or did you add in that service id filter also?

Try removing that filter and running scan without that service id. And if scanning returns results, you can try using the lower case version of the service uuid, aka `48bbe7d5-81e5-46a4-b3ad-23417934d96d`.

Also good to check that the service uuid is correct.

anwarhamr, 2020-08-28:
ACCESS_COARSE_LOCATION is not necessary if you have ACCESS_FINE_LOCATION since Fine is a higher level of permission... But I have tried with both and only left in FINE.

ahh but... let me try the lower-case thing for the service id.  I remember having to change it to uppercase in a previous release or to make it work for IOS... something like that.  I'll try w/o and w/lowercase and update
@kiafaldorius  thanks for your thoughts and help.

Brianop, 2020-09-19:
@skicson Your answer worked, thanks.

HKSOLDIER, 2020-10-09:
After the BleManager.scan() , how to get the data from the scan

anwarhamr, 2020-10-09:
@HKSOLDIER you wire-up to the event listener.  Here is snippets of mine:

```
this.handlerDiscover = bleManagerEmitter.addListener('BleManagerDiscoverPeripheral', (d) => {
           ...
        });
        this.handlerStop = bleManagerEmitter.addListener('BleManagerStopScan', () => {
          ...
        });
        this.handlerDisconnect = bleManagerEmitter.addListener('BleManagerDisconnectPeripheral', (data) => {
           ...
        });
        this.handlerUpdate = bleManagerEmitter.addListener(
            'BleManagerDidUpdateValueForCharacteristic',
            (data) => {
                ...
            },
        );
        this.handlerUpdateState = bleManagerEmitter.addListener('BleManagerDidUpdateState', async (onOff) => {
           if (onOff == 'off') {
                _this.bleActive = _this.started == false;
                _this.scanning = null;
            } else _this.bleActive = _this.started == true;

          ...
        });
```

HKSOLDIER, 2020-10-10:
Sir i didn't get it sir, please give a sample code and those UUID's i
didn't that meaning, please help  me sir

hongweiCui, 2021-03-03:
For RN Application Debug Mode
You must enable settings->location rights for the debugging APP manually in your phone's setting panel.

josethz00, 2022-08-18:
I solved this issue by doing some configurations on my code.

At the BLE Start module, I changed to this:
```diff
-BLE.start({showAlert: true}).then(() => {
+BLE.start({showAlert: true, forceLegacy: true}).then(() => {
  console.log('Módulo inicializado');
})
```

At **AndroidManifest.xml** I just updated the `maxSdkVersion` property to 33 (Android 13+)
```xml
    <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="33" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="33" />

    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="33"/>
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="33"/>
    <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
```

And at **android/build.gradle**, I changed `compileSdkVersion` and `targetSdkVersion` to 33:
```gradle
compileSdkVersion = 33
targetSdkVersion = 33
```