innoveit/react-native-ble-manager #648: On Android the peripheral.name is wrong

gragsie, 2020-07-18:
### Version

Tell us which versions you are using:

- react-native-ble-manager 7.3.0
- react-native 0.61.5
- Android 9.0

### Expected behaviour

The peripheral.name should be the Generic access device name, not the localName.
(It is correct on iOS)

### Actual behaviour

console.log of the peripheral in the BleManagerDiscoverPeripheral event:

Peripheral:  {"advertising": {"isConnectable": true, "localName": "My Device", "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "AgEGBUJsaW5rBRJ4ALQAAgoIDAlNeSBnZWxhdXRvAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {}, "serviceUUIDs": [], "txPowerLevel": 8}, "id": "0C:0B:0A:03:02:01", "name": "My Device", "rssi": -63}
### Steps to reproduce

On android, just scan for peripherals and look at the peripheral data in the BleManagerDiscoverPeripheral event

kiafaldorius, 2020-07-22:
These values come from the underlying native library calls. I personally think using the localName from the advertising as Android is doing is more correct.

https://developer.apple.com/documentation/corebluetooth/cbperipheral/1519029-name?language=objc

On ios, this can return either the advertising name or the GAP name. If it's a newly detected device that hasn't been connected to (and thus hasn't had a chance to retrieve the GAP name), the name will be the name in the advertisement. After connecting, it'll use the GAP name as long as that name is in the device cache.

Which means, even after you change the the GAP name, ios might not see the updated service name until the next connection (or you otherwise employ the service changed characteristic to tell it that the name has changed).

franzejr, 2021-04-06:
The Peripheral in this case is exactly what comes from the native library, right ?
On the example app peripheral has `.connected`: https://github.com/innoveit/react-native-ble-manager/blob/138bfac8dda1f0024f612f84212a429f36cd62ed/example/App.js#L61

But the peripheral types doesn't have the connected as attribute/function: https://github.com/innoveit/react-native-ble-manager/blob/master/index.d.ts#L2-L7

closed 2023-12-05 by marcosinigaglia