innoveit/react-native-ble-manager #653: Unhandled Promise Rejection: Peripheral did disconnect

elliottkember, 2020-08-12:
### Version

Tell us which versions you are using:

- react-native-ble-manager v7.3.1
- react-native v0.62.2
- iOS v13.6

### Expected behaviour

I'm getting crash reports that I can't seem to replicate - it seems as though a peripheral is disconnecting.

### Actual behaviour

```
Unhandled Promise Rejection: Peripheral did disconnect: 52138699-5E50-9224-F2EC-6125D841D54B
```

Unfortunately I don't get any logs other than this. It seems to just crash the app from an Objective-C exception that isn't handled in JavaScript. I'm still trying to replicate this in a debuggable version of the app, I just get these exceptions from Sentry and can't find any references to them in the list!

Do I need to check whether the device is connected before sending it a message?

kiafaldorius, 2020-08-16:
> Do I need to check whether the device is connected before sending it a message?

It's a good idea to confirm connection state before writing, because the device you're connected to can disconnect any time or there's a bluetooth link issue like the device went out of range. It's an even better idea to handle the promise rejection.

To debug this and handle the rejection, catch the error. The commands from this library returns a promise, so you can tap the promise to catch the error and see when it happened. (code written up sloppily, just here for the idea)

eg, async pattern:

```
async function sendHello(){
  try {
     await BleManager.write(...);
  } catch(e){
    console.log('write failed', e);
  }
}
```

eg, promise then-catch:

```
function sendHello(){
  BleManager.write(...).then(data => { console.log('success', data); } ).catch(e => { console.log('write failed', e) })
}
```

elliottkember, 2020-08-17:
Thanks for your reply!

> To debug this and handle the rejection, catch the error.

I already have a `.catch` on my `.write` call. So I don't think this is where the error happens - but I can't seem to get a decent stack trace in Sentry as this seems to be a native error. 

[image]

kiafaldorius, 2020-08-17:
I would check all the calls to BleManager methods. The native code doesn't do much unless requested by the Javascript interface.

The `[native code]` isn't an indication that it's from native,, the `p` and `value` certainly looks like minified bundle variable names. You may need to check your raven version and sentry setup. Maybe you need a sourcemap?

elliottkember, 2020-08-17:
Yes, I think you're right there. I've just updated the Sentry SDK, which has improved the stack traces immensely, and I pushed a build with more `.catch`es to try and catch more exceptions. I'll keep an eye on it, and I'll close this issue for now. Thanks so much!

closed 2020-08-17 by elliottkember

Burzo, 2022-08-03:
@elliottkember I'm noticing the same behavior on my end after replacing Crashlytics with Sentry, did you perhaps find a reason for the warning? I'm fairly certain all my BleManager calls are wrapped in a try/catch.

I didn't notice any problems so far - but the warning itself is pretty annoying and I don't wish to be blindsided by something in the future.

Thank you!