innoveit/react-native-ble-manager #654: [Question] Can iOS keep bluetooth connection in background?

riseworld27, 2020-08-12:
I want to know if iOS can keep bluetooth connection in background.

SAAHIIIL, 2020-08-26:
Yes. It is possible to keep the BLE Connection alive in the background as well. You check out the code and try it out.

riseworld27, 2020-08-26:
Is it supported in this library by default?  Could you indicate some part of code to demonstrate background working of BLE?

willluk, 2020-09-04:
Interested also.  We have added a listener for BleManagerModule NativeEventEmitter.  However, seems like we are not getting any read event when the app goes into background.

SAAHIIIL, 2021-01-26:
@riseworld27  The connection would be alive when alive when in background, but your reads and writes would fail in that case, you would have to do reads and writes when the app is open.

But ble manager does have such methods but you can check this out.

https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html#//apple_ref/doc/uid/TP40013257-CH7-SW10