innoveit/react-native-ble-manager #655: BleManagerDiscoverPeripheral not working with SDKVersion to 29 (First Upload to PlayStore)

Trosdan, 2020-08-13:
### Version

Tell us which versions you are using:

- react-native-ble-manager v7.3.1
- react-native v0.61.5
- Android v10

### Expected behaviour

Tell us what should happen:
When starting the device scan, the 'BleManagerDiscoverPeripheral' listener is called with the available devices.

### Actual behaviour

Tell us what happens instead
The 'BleManagerDiscoverPeripheral' listener is not being called when starting the scan

### Steps to reproduce

1.Update SDKVersion to 29 or using the latest React Native version

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

bertknliu, 2020-08-14:
I had this exact issue this morning. I did some research I did two things to solve the problem:

1) Apart from the 
    <uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" /> 
that the documentations told us to add in the Android.manifest, I also added 
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" /> 

2) I added the following code before I did anything related to Bluetooth(e.g. start, scan).
PermissionsAndroid.request(
      PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
      {
        title: 'Bluetooth Permission',
        message: 
          'In the next dialogue, Android will ask for permission for this ' +
          'App to access your location. This is needed for being able to ' +
          'use Bluetooth to scan your environment for peripherals.',
        buttonPositive: 'OK'
        },
    )

Trosdan, 2020-08-14:
Thank you very much, I was requesting `ACCESS_COARSE_LOCATION` which did not work when upgrading the SDK to 29, when changing to` ACCESS_FINE_LOCATION` the scan was working again.

closed 2020-08-18 by Trosdan