innoveit/react-native-ble-manager #656: Type System - PeripheralInfo Object Data Type is empty

amor87, 2020-08-14:
### Version

Current Version

### Expected behaviour

The type declaration for PeripheralInfo is well defined or at least decent enough to include the basic data type for that object

### Actual behaviour

It has no definition
https://github.com/innoveit/react-native-ble-manager/blob/master/index.d.ts#L120

Marvedog, 2021-06-08:
@marcosinigaglia Do I understand it correctly that it is just an `Array<string>` containing services? I extrapolate from the `retrieveServices` function and the code below.

https://github.com/innoveit/react-native-ble-manager/blob/master/android/src/main/java/it/innove/BleManager.java#L383

marcosinigaglia, 2021-06-10:
@Marvedog yes, is correct.

Marvedog, 2021-06-14:
@marcosinigaglia #747  ?

closed 2021-06-14 by marcosinigaglia