innoveit/react-native-ble-manager #662: Is it possible to do the same about the react-native-ble-lx?

vagnerwentz, 2020-09-08:
I have a question, I'm not really enjoying react-native-ble-plx, their documentation is not very good ... And I would like to know if I can scan Bluetooth, connect with the module that has a specific name, be able to send and receive data using react-native-ble-manager

marcosinigaglia, 2020-09-17:
I never used the library, I think is similar.

vagnerwentz, 2020-09-17:
> I never used the library, I think is similar.

Hi Marcos, how are you? I tried to use with TypeScript but the example that the library gives I did not understand and do not have the option to clone.

RRMoelker, 2021-01-13:
We're in the process of switching out `react-native-ble-plx` to this because the other library does not return the connected devices for Android (bug).

So far this library is able to match the features of the other library. And it works very well.

Some notes:
* There a quite a few silent failures in this library. So error handling is harder. e.g.:
  * Starting scan when Bluetooth is off.
  * Sending large data packets without specifying max bytes silently failed in our case.
* There isn't a notion of a "Device" such as in the other library. You'll need to pass the device id everywhere yourself. It's nice not having two ways to do the same thing.
* The scan is for a limited duration (no need to add that yourself)
* The TypeScript typings aren't always accurate it seems.
* If you end up using this, don't forget to write Arrays, not Uint8Arrays. The latter will fail unexpectedly.

vagnerwentz, 2022-03-14:
I done what I wanted

closed 2022-03-14 by vagnerwentz