innoveit/react-native-ble-manager #663: Cannot read name from GAP service on IOS

sessionm21, 2020-09-10:
### Version

Tell us which versions you are using:

- react-native-ble-manager v7.3.1
- react-native v0.62.2
- iOS v13.4.1

### Expected behaviour

Reading the device name from the GAP service should yield the device's name. 

### Actual behaviour

Logs show that the "Device Infomation" service is found, though for some reason the service cannot be found by UUID.

### Steps to reproduce

1. Have scanned device id
2. Await Stop scanning
3. Await Connect to device
4. Await Retrieve services
5. Try reading name from GAP service using ```BleManager.read()```

### Stack trace and console log

```
2020-09-09 20:10:20.339809-0700 app_name[1682:834260] Services Discover
2020-09-09 20:10:20.340487-0700 app_name[1682:834260] Service F3641400-00B0-4240-BA50-05CA45BF8ABC <CBService: 0x2825a1400, isPrimary = YES, UUID = F3641400-00B0-4240-BA50-05CA45BF8ABC>
2020-09-09 20:10:20.341798-0700 app_name[1682:834260] Service Device Information <CBService: 0x2825a3a80, isPrimary = YES, UUID = Device Information>
```

```
2020-09-09 20:10:20.364073-0700 app_name[1682:834430] Could not find service with UUID 00001800-0000-1000-8000-00805F9B34FB on peripheral with UUID DD413309-DEA1-2760-D2CF-FFCE351CC613
```

### Current Workaround
Added platform specific code so when ```Platform.OS === 'android'``` I read the GAP service and for IOS I use ```.name``` from the appropriate scan result.

marcosinigaglia, 2020-09-17:
Hi, can you investigate? Do you think is a library or platform bug?

sessionm21, 2020-09-18:
```
Service Device Information <CBService: 0x2825a3a80, isPrimary = YES, UUID = Device Information>
```
I have no experience with IOS development, though I think it might have something to do with how the service data is serialized into strings. I'm assuming based on the above logs that 'Device Information' should instead be the UUID in string form.

In CBPeripheral+Extensions.m there is a function ```serviceAndCharacteristicInfo``` which has mutiple references to converting things into strings.

I also just realized that I cannot write to the service & characteristic, so I'll continue to look into this.

sessionm21, 2020-09-24:
After much debugging of the library I ended up finding that IOS restricts access to the GAP service. So this is a platform specific problem.

https://devzone.nordicsemi.com/f/nordic-q-a/28811/do-i-need-to-add-the-gap-service-0x1800-to-my-application---to-change-device-name

Marking as closed.

closed 2020-09-24 by sessionm21