innoveit/react-native-ble-manager #665: null is not an object (evaluating 'bleManager.start')

joshivaibhavs, 2020-09-11:
### Version

Tell us which versions you are using:

- react-native-ble-manager v7.3.1
- expo v38.0.8
- Android v8.1

### Expected behaviour

bleManager.start() should initialize the BLE module

### Actual behaviour

throws error: null is not an object (evaluating 'bleManager.start')

### Steps to reproduce

1. Create a new project with expo
2. npm i react-native-ble-manager
3. Insert following in the main app.js file: 
`
useEffect(() => {
    bleManager.start().then(() => console.log('BLE Module Initialized')).catch(err => console.log(err))
}, [])
`

### Stack trace and console log

```
null is not an object (evaluating 'bleManager.start')
* http://192.168.1.103:19001/node_modules/expo/AppEntry.bundle?platform=android&dev=true&minify=false&hot=false:134291:20 in <unknown>
- node_modules/promise/setimmediate/core.js:45:6 in tryCallTwo
- node_modules/promise/setimmediate/core.js:200:22 in doResolve
- node_modules/promise/setimmediate/core.js:66:11 in Promise
* http://192.168.1.103:19001/node_modules/expo/AppEntry.bundle?platform=android&dev=true&minify=false&hot=false:134286:26 in start
* App.js:11:12 in useEffect$argument_0
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:18008:31 in commitHookEffectList
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:18061:28 in commitPassiveHookEffects
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:286:4 in invokeGuardedCallbackImpl
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:497:2 in invokeGuardedCallback
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:21577:27 in flushPassiveEffectsImpl
- node_modules/scheduler/cjs/scheduler.development.js:818:23 in unstable_runWithPriority
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:20435:23 in performSyncWorkOnRoot
* [native code]:null in performSyncWorkOnRoot
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:5703:31 in runWithPriority$argument_1
- node_modules/scheduler/cjs/scheduler.development.js:818:23 in unstable_runWithPriority
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:5698:21 in flushSyncCallbackQueueImpl
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:5686:28 in flushSyncCallbackQueue
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:19845:30 in scheduleUpdateOnFiber
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:23608:14 in updateContainer
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:24593:19 in ReactNativeRenderer.render
- node_modules/react-native/Libraries/ReactNative/renderApplication.js:54:4 in renderApplication
- node_modules/react-native/Libraries/ReactNative/AppRegistry.js:117:25 in runnables.appKey.run
- node_modules/react-native/Libraries/ReactNative/AppRegistry.js:213:4 in runApplication
- node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:425:19 in __callFunction
- node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:112:6 in __guard$argument_0
- node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:373:10 in __guard
- node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:111:4 in callFunctionReturnFlushedQueue
* [native code]:null in callFunctionReturnFlushedQueue
```

joshivaibhavs, 2020-09-11:
I can only find a bleManager object when I console.log my import:
```
bleManager {
  "isPeripheralConnected": [Function bound isPeripheralConnected],
}
```

marcosinigaglia, 2020-09-17:
Can you use external module in Expo?

maskaravivek, 2020-09-18:
I am having a similar issue. I ejected expo from the app but am still getting these errors. 

```
 W/ReactNativeJS: Possible Unhandled Promise Rejection (id: 0):
    TypeError: null is not an object (evaluating 'bleManager.start')
    http://localhost:8081/index.delta?platform=android&dev=true&minify=false:108212:21
    tryCallTwo@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:24385:9
    doResolve@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:24549:25
    Promise@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:24408:14
    start@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:108207:27
    http://localhost:8081/index.delta?platform=android&dev=true&minify=false:107644:43
    commitHookEffectList@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:18401:38
    commitPassiveHookEffects@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:18428:29
    invokeGuardedCallbackImpl@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:8551:21
    invokeGuardedCallback@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:8642:42
    commitPassiveEffects@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:19842:36
    commitPassiveEffects@[native code]
    wrapped@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:37715:45
    flushPassiveEffects@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:19891:32
    renderRoot@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:20353:28
    performWorkOnRoot@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:21146:23
    performWork@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:21073:30
    performSyncWork@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:21049:20
    requestWork@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:20947:26
    scheduleWork@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:20820:22
    scheduleRootUpdate@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:21314:21
    render@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:21844:26
    renderApplication@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:73759:59
    run@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:73436:28
    runApplication@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:73486:28
    __callFunction@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:2587:49
    http://localhost:8081/index.delta?platform=android&dev=true&minify=false:2344:31
    __guard@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:2541:15
    callFunctionReturnFlushedQueue@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:2343:21
    callFunctionReturnFlushedQueue@[native code]
```

It would be great if someone could point me to what might be the issue.

marcosinigaglia, 2020-09-23:
I think is not installed correctly.

maskaravivek, 2020-09-23:
Yes, the issue was that the library was not installed properly. For anyone else facing this issue, make sure that you have installed all the dependencies and have made all the config changes mentioned in the docs

closed 2020-09-24 by marcosinigaglia

Slurpgoose, 2020-11-01:
@maskaravivek do you have any advice on how to make this work? 
Ive gone through the install process three times and still have had no luck.
after each attempt I deleted package.json & node_modules/
1. `expo eject`
2. `npm i --save react-native-ble-manager`
3. update android manifest & ios plist 
4. ran `pod install` in the ios directory

 even though the library supports auto-linking for my second attempt I ran `react-native link react-native-ble-manager`.

Space-Around, 2021-01-23:
I repeated the installation of the library 4 times and everything was unsuccessful, I double-checked everything, I do not use expo

StrongLeo, 2021-12-27:
I got the same error, when running in expo.
@joshivaibhavs Do you have any ideas and solutions for this issue? Please let me know, thank you.

null is not an object (evaluating 'bleManager.scan')
at http://192.168.31.248:19000/node_modules%5Cexpo%5CAppEntry.bundle?platform=android&dev=true&hot=false&minify=false:154891:20 in <unknown>
at node_modules\react-native\node_modules\promise\setimmediate\core.js:45:6 in tryCallTwo
at node_modules\react-native\node_modules\promise\setimmediate\core.js:200:22 in doResolve
at node_modules\react-native\node_modules\promise\setimmediate\core.js:66:11 in Promise
at node_modules\react-native-ble-manager\BleManager.js:246:47 in scan
at components\Printing.js:16:25 in startScan
at components\Printing.js:202:33 in TouchableOpacity.props.onPress
at node_modules\react-native\Libraries\Pressability\Pressability.js:697:17 in _performTransitionSideEffects
at node_modules\react-native\Libraries\Pressability\Pressability.js:634:6 in _receiveSignal
at node_modules\react-native\Libraries\Pressability\Pressability.js:528:8 in responderEventHandlers.onResponderRelease
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:93:4 in invokeGuardedCallbackProd
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:318:2 in invokeGuardedCallback
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:342:2 in invokeGuardedCallbackAndCatchFirstError
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:439:41 in executeDispatch
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:463:19 in executeDispatchesInOrder
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:2563:28 in executeDispatchesAndRelease
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:1212:4 in forEachAccumulated
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:2588:20 in runEventsInBatch
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:2735:18 in runExtractedPluginEventsInBatch
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:2681:35 in batchedUpdates$argument_0
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:18204:13 in batchedUpdates$1
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:2534:29 in batchedUpdates
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:2680:16 in _receiveRootNodeIDEvent
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:2800:27 in receiveTouches
at node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:414:4 in __callFunction
at node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:113:6 in __guard$argument_0
at node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:365:10 in __guard
at node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:112:4 in callFunctionReturnFlushedQueue

mikasmart, 2022-02-02:
I've the same error.
Have you find a solution ?

StrongLeo, 2022-02-02:
Sorry, not yet. In fact, I have give up using this package.

elukking, 2022-03-03:
Y Ahora que package estas usando  StrongLeo

mikasmart, 2022-03-04:
Don't use expo; i use only react-native without expo and it's work.

JamesOwen7, 2022-04-17:
Has anyone resolved this?

JamesOwen7, 2022-04-21:
> add `restoreIdentifierKey` string to your initiation
> 
> BleManager.start({ showAlert: false, restoreIdentifierKey: 'xxxxxxxxxxxx' })

I still get the same error

Psalmzz, 2022-04-22:
have you added the ble_manager to your pod file and ran `pod install`

  pod 'react-native-ble-manager', :path => '../node_modules/react-native-ble-manager'

JamesOwen7, 2022-04-22:
I'm on a windows machine and was told pod install wasn't required...?

JamesOwen7, 2022-04-22:
But yes the package does sit within node_modules

Psalmzz, 2022-04-22:
I guess you are building for android then I will suggest you check your permission set

briandilley, 2022-04-23:
I have the same issue, here's my stuff:
```
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
          xmlns:tools="http://schemas.android.com/tools"
  package="com.trinketmanager">

    <uses-permission android:name="android.permission.INTERNET" />

    <uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT"  />

    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    <uses-permission-sdk-23 android:name="android.permission.ACCESS_FINE_LOCATION" />

    <!-- Add this line if your application always requires BLE. More info can be found on:
        https://developer.android.com/guide/topics/connectivity/bluetooth-le.html#permissions
      -->
    <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>

    <application
      android:name=".MainApplication"
      android:label="@string/app_name"
      android:icon="@mipmap/ic_launcher"
      android:roundIcon="@mipmap/ic_launcher_round"
      android:allowBackup="false"
      android:theme="@style/AppTheme">
      <activity
        android:name=".MainActivity"
        android:label="@string/app_name"
        android:configChanges="keyboard|keyboardHidden|orientation|screenLayout|screenSize|smallestScreenSize|uiMode"
        android:launchMode="singleTask"
        android:windowSoftInputMode="adjustResize"
        android:exported="true">
        <intent-filter>
            <action android:name="android.intent.action.MAIN" />
            <category android:name="android.intent.category.LAUNCHER" />
        </intent-filter>
      </activity>
    </application>
</manifest>
```

```
// Top-level build file where you can add configuration options common to all sub-projects/modules.

buildscript {
    ext {
        buildToolsVersion = "31.0.0"
        minSdkVersion = 16
        compileSdkVersion = 31
        targetSdkVersion = 29
        ndkVersion = "21.4.7075529"
        ffmpegKitPackage = "full-lts"
    }
    repositories {
        google()
        mavenCentral()
    }
    dependencies {
        classpath("com.android.tools.build:gradle:7.0.4")
        classpath("com.facebook.react:react-native-gradle-plugin")
        classpath("de.undercouch:gradle-download-task:4.1.2")
        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url("$rootDir/../node_modules/react-native/android")
        }
        maven {
            // Android JSC is installed from npm
            url("$rootDir/../node_modules/jsc-android/dist")
        }
        mavenCentral {
            // We don't want to fetch react-native from Maven Central as there are
            // older versions over there.
            content {
                excludeGroup "com.facebook.react"
            }
        }
        google()
        maven { url 'https://www.jitpack.io' }
    }
}
```