innoveit/react-native-ble-manager #669: We in our team have two Android 10 devices, none of them can detect devices on them using react-native-ble-manager

abhinavsiotech, 2020-10-01:
We in our team have two Android 10 devices, none of them can detect devices on them using react-native-ble-manager.

The code is as is as in your Git ReadMe.

Package.json:

{
  "name": "vm_configuration_app",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "start": "react-native start",
    "test": "jest",
    "lint": "eslint .",
    "postinstall": "jetify",
    "reset": "yarn cache clean && react-native start -- --reset-cache",
    "and_bundle": "react-native bundle --platform android --dev false --entry-file index.js --bundle-output android/app/src/main/assets/index.android.bundle --assets-dest android/app/src/main/res"
  },
  "dependencies": {
    "@aws-amplify/auth": "^3.3.6",
    "@react-native-community/art": "^1.2.0",
    "@react-native-community/async-storage": "^1.11.0",
    "@react-native-community/masked-view": "^0.1.10",
    "@react-native-community/netinfo": "^5.9.6",
    "amazon-cognito-identity-js": "^4.3.5",
    "aws-amplify": "^3.0.25",
    "aws-amplify-react-native": "^4.2.5",
    "convert-string": "^0.1.0",
    "react": "16.13.1",
    "react-native": "0.63.2",
    "react-native-android-wifi": "0.0.41",
    "react-native-base64": "^0.1.0",
    "react-native-ble-manager": "^7.3.1",
    "react-native-elements": "^2.2.1",
    "react-native-gesture-handler": "^1.7.0",
    "react-native-linear-gradient": "^2.5.6",
    "react-native-progress": "^4.1.2",
    "react-native-reanimated": "^1.13.0",
    "react-native-safe-area-context": "^3.1.4",
    "react-native-screens": "^2.10.1",
    "react-native-vector-icons": "^7.0.0",
    "react-navigation": "^4.4.0",
    "react-navigation-stack": "^2.8.2"
  },
  "devDependencies": {
    "@babel/core": "^7.11.1",
    "@babel/runtime": "^7.11.2",
    "@react-native-community/eslint-config": "^2.0.0",
    "babel-jest": "^26.3.0",
    "eslint": "^7.6.0",
    "jest": "^26.4.0",
    "metro-react-native-babel-preset": "^0.62.0",
    "react-test-renderer": "16.13.1"
  },
  "jest": {
    "preset": "react-native"
  },
  "rnpm": {
    "assets": [
      "./assets/fonts"
    ]
  }
}
React-Native info

react-native info
info Fetching system and libraries information...
(node:6348) Warning: Accessing non-existent property 'padLevels' of module exports inside circular dependency
(Use node --trace-warnings ... to show where the warning was created)
System:
OS: macOS 10.15.6
CPU: (4) x64 Intel(R) Core(TM) i5-3210M CPU @ 2.50GHz
Memory: 47.28 MB / 8.00 GB
Shell: 5.7.1 - /bin/zsh
Binaries:
Node: 14.8.0 - /usr/local/bin/node
Yarn: 1.22.4 - /usr/local/bin/yarn
npm: 6.14.7 - /usr/local/bin/npm
Watchman: Not Found
Managers:
CocoaPods: 1.9.3 - /usr/local/bin/pod
SDKs:
iOS SDK:
Platforms: iOS 14.0, DriverKit 19.0, macOS 10.15, tvOS 14.0, watchOS 7.0
Android SDK:
API Levels: 23, 29, 30
Build Tools: 23.0.1, 28.0.3, 29.0.2, 29.0.3, 30.0.1
System Images: android-30 | Google APIs Intel x86 Atom
Android NDK: Not Found
IDEs:
Android Studio: 4.0 AI-193.6911.18.40.6626763
Xcode: 12.0.1/12A7300 - /usr/bin/xcodebuild
Languages:
Java: 1.8.0_265 - /usr/bin/javac
Python: 2.7.16 - /usr/bin/python
npmPackages:
@react-native-community/cli: Not Found
react: 16.13.1 => 16.13.1
react-native: 0.63.2 => 0.63.2
npmGlobalPackages:

marcosinigaglia, 2020-10-01:
Hi, probably is a permission issue.

abhinavsiotech, 2020-10-01:
No It is working on Android 8 devices.

kapernikxd, 2020-10-21:
Same problem. In Android 10 does not work, but in Android 6 it's fine.

WookieFPV, 2020-10-22:
I had the same issue, take a look at PR #682 (this will solve the android 10 issue)

andandrej, 2020-11-03:
After adding permissions: ACCESS_FINE_LOCATION and ACCESS_COARSE_LOCATION in AndroidManifest.xml nothing changes. After scanning it returns [], what should I do?

andandrej, 2020-11-03:
I fixed this issue with [this comment](https://github.com/innoveit/react-native-ble-manager/issues/647#issuecomment-682884881). Problem was that I had turned off location on phone. In AndroidManifest.xml I have:
- BLUETOOTH
- BLUETOOTH_ADMIN
- ACCESS_FINE_LOCATION
- ACCESS_COARSE_LOCATION

Hope this will help you too! :)

closed 2020-11-18 by marcosinigaglia

abssaxon, 2020-12-01:
> I fixed this issue with [this comment](https://github.com/innoveit/react-native-ble-manager/issues/647#issuecomment-682884881). Problem was that I had turned off location on phone. In [...]

These permissions are already documented. We have already used them.