innoveit/react-native-ble-manager #674: BleManager.createBond got 2 arguments, expected 3 - this error popups when I try for auto bonding

siddaiahm, 2020-10-06:
### Before open an issue
- Check the closed issues, your question maybe is not new.
- We can't debug your physical devices.
- If the library is not working 99% you did something wrong, in the code or in the phone permissions.

### Version

Tell us which versions you are using:

- react-native-ble-manager v?.?.?
- react-native v?.?
- iOS/Android v.?

### Expected behaviour

Tell us what should happen

### Actual behaviour

Tell us what happens instead

### Steps to reproduce

1.
2.
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

siddaiahm, 2020-10-06:
I have made a pull request on this, Sorry for my mistake in previous pull request.

marcosinigaglia, 2020-10-06:
Hi, i released a patch, please be more careful, I can't test without a device.

closed 2020-10-06 by marcosinigaglia

siddaiahm, 2020-10-06:
it is working perfect, if i take instance  from the NativeModules.BleManager 
eg : 
const bleManager = NativeModules.BleManager;
 bleManager.createBond(peripheralId, peripheralPin, (error) => {
      if (error) {
        reject(error);
      } else {
        fulfill();
      }
    });
this code works perfect
but when i try to use 
import bleManager from "react-native-ble-manager";
bleManager.createBond (peripheralId, pin).then(()=>{}).catch(()=>{});

this still gives me the same error. please check on this i am not able to figure out the issue.