innoveit/react-native-ble-manager #686: Retrieve Service UUID and Characteristic UUID

benjaminle9x, 2020-11-18:
Tell us which versions you are using:

- react-native-ble-manager v7.4.1
- react-native v0.63.3
- Android version 10

I'm testing the ESP32 module (BLE module) on my React Native app and it requires some texts to be written over the BLE. I used the retrieveServices() function from the tutorial but I do not know how to retrieve the service UUID and characteristic UUID dynamically. How do I do this?

marcosinigaglia, 2020-11-18:
Hi, what do you mean by dynamically?

benjaminle9x, 2020-11-19:
> Hi, what do you mean by dynamically?

What I mean is how do I access the serviceUUID and characteristicUUID property instead of console logging peripheralInfo, and then hard code it in my app.

marcosinigaglia, 2020-11-19:
If you have to hardcode in the app I don't understand why you can't use simply the console logging. You can change the example code and show the services and characteristics list.

benjaminle9x, 2020-11-19:
> If you have to hardcode in the app I don't understand why you can't use simply the console logging. You can change the example code and show the services and characteristics list.

Well it won't be convenient if I have like 10 different BLE modules and each time, when I want to send something to a different module, I have to go to the code and change the servUUID and characteristicUUID. Instead, is there any way I can do like using peripheralInfo.service.UUID or something like that?

marcosinigaglia, 2020-11-19:
When you call retrieveServices you have the services and characteristics list so you can do what you need.

closed 2020-11-19 by marcosinigaglia