innoveit/react-native-ble-manager #689: is support bluetooth gps ？

zhangyuwu123, 2020-11-26:
### Before open an issue
- Check the closed issues, your question maybe is not new.
- We can't debug your physical devices.
- If the library is not working 99% you did something wrong, in the code or in the phone permissions.

### Version

Tell us which versions you are using:

- react-native-ble-manager v?.?.?
- react-native v?.?
- iOS/Android v.?

### Expected behaviour

Tell us what should happen

### Actual behaviour

Tell us what happens instead

### Steps to reproduce

1.
2.
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

marcosinigaglia, 2020-11-30:
I don't understand, gps?

closed 2020-12-16 by marcosinigaglia

roysG, 2023-01-14:
@Symous, 
Absolutely right, so stupid requirement!

Any update on this?

marcosinigaglia, 2023-01-16:
Is not a library requirement but an Android requirement.