innoveit/react-native-ble-manager #690: Invariant Violation: Module AppRegistry is not a registered callable module (calling runApplication) on iOS devices

Girum, 2020-11-27:
Hi.

I used the example and modified it. But, when I tried to run on iOS device, it says:

```
  Invariant Violation: Module AppRegistry is not a registered callable module (calling runApplication)
[Fri Nov 27 2020 08:54:22.670]  ERROR    Invariant Violation: Module AppRegistry is not a registered callable module (calling runApplication)
```

I realized that the NativeModules and NativeEventEmitter was the problem. I removed them and it starts to work... but nothing happens. I saw a solution on the web tu uso NativeAppEventEmitter, but it dont work too.

How cal I solve this?

Note: in Android it works fine

Girum, 2020-11-27:
I verified and use the NativeAppEventEmmiter. The android continous goes Ok. The iOS receive the devices, but only in the Xcode terminal

marcosinigaglia, 2020-11-30:
I think is some installation issue.

voulgarakis, 2021-04-23:
cd ios
pod install
cd ..
react-native run-ios

that fixed the issue for me. 

@marcosinigaglia please update the documentation for iOS

closed 2021-04-23 by marcosinigaglia