innoveit/react-native-ble-manager #692: Pairing Success CallBack issue

HCLThiruG, 2020-12-08:
we are trying to pairing the iOS and Android device with BLE device but unable get the success of failure callback. 

Please suggest us how to get the Pairing success or failure callback in both iOS and Android.

franzejr, 2021-04-21:
as far as I see the only thing this library supports is if we have a connection.

cfradella, 2023-12-12:
On Android you can set an event emitter listener for the event `BleManagerConnectPeripheral` to know if you paired. However, the concept of "bonding" is only an Android concept, which is how you'll know if the bluetooth device has been save to memory for future use.

Android can use the same method above, but for the event `BleManagerPeripheralDidBond`, however I don't know of a way right now, using this library, if we can check if the paired device was saved on iOS. i.e., pressing the pairing OS modal has no event to listen to for success/fail.

closed 2024-03-19 by marcosinigaglia