innoveit/react-native-ble-manager #694: Android 10 - startNotification gives error "Error writing descriptor stats=133"

viktorbk, 2020-12-11:
### Version
- react-native-ble-manager v7.4.1
- react-native v0.63.4
- iOS/Android v.10

### Expected behaviour
startNotification should start notification on characteristic

### Actual behaviour
Gives error: Error writing descriptor stats=133

### Code to reproduce
AndroidManifest.xml
```
    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
```
App.js
```
  const connect = async device => {
    await BleManager.connect(device.id)
    console.log('Connected......')
    const info = await BleManager.retrieveServices(device.id)
    console.log('Got service...', info)
    const characteristic = info.characteristics[3]
    const service = characteristic.service
    const charect = characteristic.characteristic
    console.log(service, charect)
    setTimeout(async () => {
      await BleManager.startNotification(device.id, service, charect)
      debugger
      await BleManager.disconnect(device.id)
      console.log('Disonnected......')
    }, 500)
  }
```

### Stack trace and console log
```
{advertising: {…}, rssi: -56, id: "04:91:62:93:03:59", name: "IDLock150"}
App.js:65 Connected......
App.js:67 Got service... {characteristics: Array(4), services: Array(3), advertising: {…}, name: "IDLock150", rssi: -56, …}
App.js:71 11020304-0506-0708-0900-0a0b0c0d0e0f 11223344-5566-7788-9900-aabbccddeeff
backend.js:32 Possible Unhandled Promise Rejection (id: 0):
"Error writing descriptor stats=133"
...

```

marcosinigaglia, 2020-12-16:
Hi, I don't have an android 10 phone so i can't  reproduce your issue.

viktorbk, 2020-12-16:
Hi, if you buy Nokia 5.3 with Android 10 as I did i will pay for the phone. There are already customers complaining since our app is in Google Play store and important functionality for our company.

SCasarotto, 2021-01-12:
I can confirm as I just ran identical code on android 10 and 11 and everything worked correctly on Android 11 but not android 10. My error was `Error writing descriptor stats=1`. @viktorbk did you ever find a solution to this?

zorba0517, 2021-06-10:
i got same issue

marcosinigaglia, 2021-06-10:
It's a OS/Hardware bug.

atornatore, 2021-06-17:
Did anybody find a workaround for this?

jimhuf, 2021-07-27:
I have this same problem, on a Samsung note 10
Android 11 (Kernel 4.14)
But works fine on Android 11 (Kernel 4.9) Galaxy Tab S5e

Any update on this issue?

DomLeMarin, 2021-08-30:
I had the same issue with the phone OPPO A53 (ColorOS 7.2 based on Android 10). After struggling on it for several days,
I found out that I need to **create a bond first** and then it works (I never had to do that with any other Android phones). 

After catching the error stats=133, the following seems to work:
1) Wait until the device is truly disconnected (about 10 sec),
2) Call the function _BleManager.createBond(peripheral.id)_,
3) Accept the pairing pop-up that shows up on the screen,
4) Wait until the device is truly disconnected (about 25 sec),
5) Start again with _BleManager.connect_, _BleManager.retrieveServices_, _BleManager.startNotification_

This needs to be done only once. The next connections are working as usual, since the two devices are bonded.

I hope it helps someone...

2sem, 2023-09-05:
the error rarely occured.
I saw it once.

closed 2023-12-04 by marcosinigaglia