innoveit/react-native-ble-manager #695: Need a restart for scanning peripherals

GalliusJulius, 2020-12-13:
Hello

### Version

- react-native-ble-manager v4.4.1
- react-native v0.63.4
- iOS/Android v. 10 QKQ1.190825.002

### Expected behaviour

Peripherals should appear after scanning (same behaviour with my app or the example)

### Actual behaviour

Peripherals are only appear once the app is installed on the phone (with react-native run-android) and I restart my phone.

### Steps to reproduce

1. react-native run-android (in the example folder) => running on a android phone throw USB.

### Stack trace and console log

It's entering in the .then of the scan method (Scanning is log in the console).
I don't see other logs that could be useful (don't hesitate to ask for some if needed)

The workaround works fine for dev but I hope I will not have the same issue in prod.

In advance thank you.

rfestag, 2020-12-30:
Can you provide a sample of what you're doing? The scan only runs of the specified period of time, and that's it. If you want to scan again (or scan continuously), you need to run another scan. Note that running the scan again will clear out the discovered peripherals, so if you want a running list of all peripherals, you'll need to maintain that state separately. 

If you are just having trouble knowing when you have peripherals, you can use the BleManagerDiscoverPeripheral event (called everytime an advertisement is seen), or you can periodically call `getDiscoveredPeripherals`

GalliusJulius, 2021-01-01:
I have this issue with the example provide in the react-native-ble-menager repository. The issue here is that BleManagerDiscoverPeripheral event isn't triggered even with peripherals available close to my phone. I don't think the issue come from my phone as I can see BLE devices with other applications (nerf connect for instance).

iAmHarmon, 2021-01-26:
I am also having this issue with Android (10). I have to disable then reenable location permissions in able to start discovering peripherals.

GalliusJulius, 2021-01-26:
I still didn't find a solution but I guess I know where it's coming from :
The location permission pop-up only asks if I want to use location only when the app is running or never but not the third option : always.
When I go to the settings of my app I can switch this permission from only when the app is running to always and it's working fine.

So lot of questions : 

- Why do I not have the always allowed proposition on my app? 
- Is there an issue with the state handle of the application in the example (in my case)?
- Why is it not working with the only allowed when running?

@iAmHarmon can you confirm this (not having the possibility to always allow localization in the user pop-up)?

rfestag, 2021-01-26:
I'll take a stab:

1. Does adding the BACKGROUND_LOCATION cause "always" to show up - https://stackoverflow.com/questions/61909313/allow-all-the-time-location-prompt-not-coming-in-android-sdk-29
2. I'm curious if you need to re-request the permission after you resume if you don't have the background permission?

GalliusJulius, 2021-01-26:
> I'll take a stab:
>
> 1. Does adding the BACKGROUND_LOCATION cause "always" to show up - https://stackoverflow.com/questions/61909313/allow-all-the-time-location-prompt-not-coming-in-android-sdk-29
> 2. [...]

At first I also thought it was that,
But I already have
  `<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION"/>`
in my manifest

franzejr, 2021-05-02:
@GalliusJulius , did you get it working ?

zhifeng-heng, 2021-07-08:
Could it be because requesting ACCESS_BACKGROUND_LOCATION permission needs to be handled differently on different Android versions?

This could help: https://medium.com/swlh/request-location-permission-correctly-in-android-11-61afe95a11ad

Rowansdabomb, 2021-08-20:
I am having this issue as well, any updates?

zhifeng-heng, 2021-08-21:
> I am having this issue as well, any updates?

Hi Rowansdaomb, unfortunately no. I am still facing this issue even now.

Can you share if it only happens on a single device on ios only, or there are other scenarios it is happening. Maybe we can find some clues to try to get to a solution / workaround.

rUCUS, 2021-12-21:
Hi, having the same issue

bassamyan, 2021-12-22:
i have this issue as well !!!

closed 2023-12-04 by marcosinigaglia