innoveit/react-native-ble-manager #696: Scan little device

batkhai290499, 2020-12-14:
someone help me. all version i use for my project is newest

here is my code
import React, { useState } from "react";
import { Container, Header, Content, Tab, Tabs, Text, Button } from 'native-base';
import { Col, Row, Grid } from 'react-native-easy-grid';
import { View, Alert, NativeAppEventEmitter, NativeModules, NativeEventEmitter } from 'react-native';
import BleManager from 'react-native-ble-manager';
import { useEffect } from "react";

const BleManagerModule = NativeModules.BleManager;
const bleManagerEmitter = new NativeEventEmitter(BleManagerModule);

function BluetoothScreen({ navigation }) {
    const [peripherals, setPeripherals] = useState([])

    useEffect(() => {

        BleManager.start({ showAlert: false })
            .then(() => {
                console.log('BleManager initialized');

                bleManagerEmitter.addListener(
                    'BleManagerDiscoverPeripheral',
                    (args) => {
                        console.log(args);

                        if (args.name !== null) {
                            var peripheral = { 'id': args.id, 'name': args.name };
                            peripherals.push(peripheral);

                            var newArr = []
                            for (var i = 0; i < peripherals.length; i++) {
                                if (newArr.indexOf(peripherals[i]) === -1) {
                                    newArr.push(peripherals[i])
                                    setPeripherals(newArr);
                                }
                            }
                        }

                    }
                );

                BleManager.checkState();
            })
            .catch((error) => {
                console.log(error);
            });

    }, [])

    const scanForDevices = () => {
        BleManager.scan([], 5, false)
            .then((args) => {
                console.log(args);
                console.log('Scan started');
                setPeripherals([]);
            })
            .catch((error) => {
                console.log(error);
            });

            
    }

    // const viewDevice = () => {
    //     BleManager.getBondedPeripherals([]).then((bondedPeripheralsArray) => {
    //         // Each peripheral in returned array will have id and name properties
    //         console.log(bondedPeripheralsArray);
    //         setPeripherals(bondedPeripheralsArray);
    //     });
    // }

    const connect = (item) => {
        console.log(item.id);
        BleManager.connect(item.id)
            .then(() => {
                // Success code
                console.log("Connected");
                setTimeout(() => {
                    BleManager.retrieveServices(item.id).then((peripheralData) => {
                        console.log('Retrieved peripheral services', peripheralData);
                        BleManager.readRSSI(item.id).then((rssi) => {
                            console.log('Retrieved actual RSSI value', rssi);
                        });
                        var characteristics = peripheralData.characteristics
                        console.log('Retrieved characteristics:', characteristics);
                    });
                }, 900);
            })
            .catch((error) => {
                // Failure code
                console.log(error);
            });

    }
    return (
        <Container>

            <Button onPress={() => scanForDevices()} >
                <Text>Scan</Text>
            </Button>
            <View>
                {/* {
                    peripherals.map((item, key) => {
                        <Text>{item.name}</Text>
                    })
                } */}
                {peripherals.map((item, key) => {
                    return (
                        // <Text key={key}>{item.name}</Text>
                        <Button onPress={() => connect(item)} key={key}>
                            <Text >{item.name}</Text>
                        </Button>
                    )
                })}
            </View>

            {/* <Button onPress={() => connect()} >
                <Text>Connect</Text>
            </Button> */}
        </Container >
    );
};

export default BluetoothScreen;

rfestag, 2020-12-30:
What help do you need? I'm not seeing a description of the problem. Most folks probably won't try to read your code to guess your intent and why this doesn't behave as you expect?

Additionally, it would be helpful to fix the code sections. Looks like some of your code listed is above and below the area treated as a code section.

batkhai290499, 2021-01-02:
thank for your help. i have a lot of idea for fix them. Thank for your lib <3
I forced to used react-native-bluetooth-classic. because your lib just scan for my app is some smart watch (bluetooth version >4.2)
my device just have a version 4.0 or lower.

closed 2021-01-08 by marcosinigaglia