innoveit/react-native-ble-manager #699: Connection: Not connected but no exception

Gueoff, 2021-01-07:
### Version

Tell us which versions you are using:

- react-native-ble-manager v7.3.1
- react-native v0.61.4
- Android v.10

### Expected behaviour

When launching BleManager.connect function, the peripheral should be connected or it should trigger an exception.

### Actual behaviour

The connect function fails but it doesn't trigger an exception.

### Steps to reproduce

1. Launch connect()
2. Launch isConnected()
3. isConnected returns false

### Stack trace and console log

I have a function which set a timeout:
```
  async launchFunction(promise) {
    clearTimeout(this.idTimeout);
    let timeout = new Promise((resolve, reject) => {
      this.idTimeout = setTimeout(() => {
        const error = {
          code: AppConstants.ERRORS.BLUETOOTH_TIMEOUT,
          method,
        };
        reject(error);
      }, AppConstants.BLUETOOTH.TIME_CONNECT);
    });

    return Promise.race([promise(), timeout]);
  }
```
  
An other function allows me to connect to the peripheral and read "retrieve services". Then I can do what I want with read/writes etc.

````
async connect(tool, listener = true) {
    const identifier = BluetoothService.getIdentifier(tool);

    // Try to connect
    await this.launchFunction(() => BleManager.connect(identifier));
    isConnected = await BleManager.isPeripheralConnected(identifier);

      if (!isConnected) {
        throw {
          code: AppConstants.ERRORS.BLUETOOTH_CONNECT,
          method: this.method,
        };
    }

    // Retrieve service to have access to read/write.
    console.log('BLE retrieveServices', identifier);
    await this.launchFunction(
      () => BleManager.retrieveServices(identifier),
      'retrieveServices',
    );
...
````

Sometimes, the connect function works fine and I am connected to my peripheral. 
Sometimes, I launch the connect function, it doesn't trigger any exception, then launch the isConnected function which return false.

marcosinigaglia, 2021-01-12:
Hi, you have to handle the connection timeout yourself.

Gueoff, 2021-01-13:
Hi marco!
Thanks for your response, but that's what I'm doing with my "launchFunction" method. It races the connect() function and a timeout.
The problem i'm facing here is that when I launch connect() then isConnected(), the isConnected() function returns false (not always).
The peripheral I try to connect to has the "isConnected" state but the phone doesn't

Gueoff, 2021-02-08:
I found a solution to my problem but maybe we can comment the documentation of the library with that:

As weird as it sounds, we highly recommend calling discoverServices() from the main/UI thread to prevent a rare threading issue from causing a deadlock situation where the app can be left waiting for the onServicesDiscovered() callback that somehow got dropped. This issue happens very rarely, and may have been fixed in recent Android versions, but since we’re supporting Android 5.0 and up, this workaround is worth it just for the peace of mind.

I am actually using redux-saga and launched connect() and retrieveServices() in a saga which create a different thread, it cause problems in android.

closed 2021-02-08 by Gueoff