innoveit/react-native-ble-manager #700: Is it possible to bring both the id and the mac address at the same time on both android and ios?

Heltonphg, 2021-01-12:
Tell us which versions you are using:

react-native-ble-manager v7.4.1
react-native v0.63.4
Android version 10

I need the peripheral to bring both the id and the mac address in the same object, it's possible?

` bleManagerEmitter.addListener(
      'BleManagerDiscoverPeripheral',
      (peripheral) => {
        console.log(peripheral)
      },
    );
`

marcosinigaglia, 2021-01-12:
The id on Android is the mac address, in iOS is an internal id.

Heltonphg, 2021-01-12:
What is the difference between mac address to internal id?

marcosinigaglia, 2021-01-12:
iOS decide to hide the mac address so they show you a generated id.

closed 2021-01-12 by marcosinigaglia

turkogluc, 2021-07-19:
Is there a way to retrieve mac address from the discovered peripheral?