innoveit/react-native-ble-manager #706: Failed to start notification

nolife08021, 2021-02-05:
"Set notification failed for 000036f6-0000-1000-8000-00805f9b34fb".I did add the time out between retrieveServices and startNotification but set notification failed too. Any solution to solve it?

nolife08021, 2021-02-05:
BleManager.start({showAlert: false})
          .then(() => {
              console.log('Module initialized');
              let id = 'FA:97:B4:F4:26:65';
              BleManager.connect(id)
                  .then(() => {
                      console.log('Connected');
                      BleManager.retrieveServices(id)
                          .then((peripheralInfo) => {
                              console.log('Peripheral info:', peripheralInfo);
                              setTimeout(function () {
                                  console.log('in timeout');
                                  BleManager.startNotification(id, 'fee7', '36f6')
                                      .then(() => {
                                          console.log('Notification success');
                                      })
                                      .catch((error) => {
                                          console.log(error);
                                      });
                              }, 3000);
                          })
                          .catch((error) => {
                              console.log(error);
                          });
                  })
                  .catch((error) => {
                      console.log(error);
                  });
          })
          .catch((error) => {
              console.log(error);
          });

heusgenda, 2021-02-18:
BleManager.startNotification(id, 'fee7', '36f6') in this method you need to swap fee7 with 36f6 (1. MAC, 2. serviceUUID, 3. characteristic)

closed 2021-02-18 by marcosinigaglia

nolife08021, 2021-02-22:
> BleManager.startNotification(id, 'fee7', '36f6') in this method you need to swap fee7 with 36f6 (1. MAC, 2. serviceUUID, 3. characteristic)

sorry, i dont understand. fee7 is the serviceUUID and 36f6 is the notify characteristic.