innoveit/react-native-ble-manager #714: iOS - BleManagerDisconnectPeripheral not called for connected peripherals when disabling device bluetooth

NeilMcGuiness, 2021-02-24:
Hi all, 

Apologies if this is a duplicate but I have searched both open and closed issues with all keywords I could think of to find this and couldn't.

### Version

- react-native-ble-manager v6.7.0
- react-native v0.61.4
- iOS v 12.1
- iOS v 14.2
- iOS v 14.4
- Works on Android 7, 8, 9 and 10

### Expected behaviour
While connected to a peripheral, switch off device bluetooth and the BleManagerEmitter should emit a 'BleManagerDisconnectPeripheral' event, with the id of the currently connected peripheral to show that it was disconnected when the bluetooth was disabled.

This works perfectly on all versions of Android we have tested. 

### Actual behaviour

On all iOS devices we have tested, the event is not emitted. Also our peripherals have a buzzer in the hardware which plays a melody when the device is connected and another melody when disconnected. When disabling device bluetooth on Android while connected, the peripheral plays the disconnect melody. However when doing the same on iOS, no melody is played so it seems the peripheral has not been disconnected.

### Steps to reproduce

1. Connect to a peripheral using iOS
2. Disable device bluetooth
3. Observe (using console.log() etc) that no BleManagerDisconnectPeripheral is fired when doing #2

marcosinigaglia, 2021-04-23:
I don't think is a library problem but how iOS handle the connection in this case. Did you tried to close the connection manually after the turning off?

Edai, 2021-07-15:
Same here ! The user manually disables the Bluetooth on iOS via the Settings menu but there is no BleManagerDisconnectPeripheral fired. Any suggestions ?

zakton5, 2023-06-28:
Any updates on this? This issue has been open for over 2 years.

marcosinigaglia, 2023-12-13:
This issue is stale because it has been open 180 days with no activity. Remove stale label or comment or this will be closed in 7 days.

SergioGCarvalho, 2023-12-18:
same problem here

closed 2023-12-21 by marcosinigaglia