innoveit/react-native-ble-manager #716: Could not find characteristic with ios device

Alexispap, 2021-03-02:
### Version

- react-native-ble-manager v7.4.2
- react-native v0.63.3
- iOS all versions

### Expected behaviour

After connecting to a peripheral, the device should use startNotification to write a value in a specific characteristic.

### Actual behaviour

Connection to the peripheral is established and the items properties are available. 

But startNotification throws the error message "Could not find characteritic with UUID xxxxxxxxx on service with UUID yyyyyyyy on peripheral with UUID zzzzzzzz"

On Android devices, everything works fine and the characteristic is available and can be written.

### Steps to reproduce

1. Starting BLE module
2. Connecting to the peripheral
3. Use startNotification 

### Stack trace and console log
Error when using startNotification:  "Could not find characteritic with UUID xxxxxxxxx on service with UUID yyyyyyyy on peripheral with UUID zzzzzzzz"

djain1992, 2021-03-02:
Did you try changing the case to lower case for char UUID and service UUID ?

Alexispap, 2021-03-02:
Yes, both are not working. I tried with another package. It is also not working. So i suspect that there is a general issue with ios here. Do i need some special permissions or do i have to configure the characteristic on the peripheral in some special way?

Alexispap, 2021-03-04:
Is this an issue that comes from passing a normal string to the API, whilst the native module wants a NSString?

marcosinigaglia, 2021-03-05:
Hi, have you call retrieveServices? Is the characteriti/service listed?

Alexispap, 2021-03-05:
Yes, i called this function. The service is listed and some characteristics, but not the one that i need. Are there special restrictions on the UUID or do i have to enable something in Xcode for it to work?

djain1992, 2021-03-05:
You can use app like light blue to connect and check all the services
present. If light blue does not show it, it means device does not have it.

Alexispap, 2021-03-05:
It exists, since on Android it works totally fine.

djain1992, 2021-03-05:
If you don’t mind sharing the screen capture for LightBlue showing services
and characteristics and console log for retrieve services for comparison, I
can check and see if any find any clue.

Alexispap, 2021-03-05:
[image]
[image]
[image]
[image]

Alexispap, 2021-03-07:
In the release build the app even crashes. @djain1992 @marcosinigaglia Do you have any idea? I have to fix it very soon 😄 Thanks for your time!!

Alexispap, 2021-03-08:
This is the console output on Android devices (here, the characteristic is shown in contrast to ios):
[image]

djain1992, 2021-03-08:
1. On IOS if you have lightblue and lightblue shows the characteristics, then we can rule out the issue with the device completely.
2. What is the exact error message you see : But startNotification throws the error message "Could not find characteritic with UUID xxxxxxxxx on service with UUID yyyyyyyy on peripheral with UUID zzzzzzzz"
3. Do you explicitly call peripheral.retrieveServices after the connect call has been success. On IOS you need to explicitly retrieve each service after the connection, Android automatically does it.

Alexispap, 2021-03-08:
The log of the Iphone in development ist:  Connected to 204B2876-CEAC-62FC-9B8E-6088E73B78CD
[Sat Mar 06 2021 14:25:02.183]  DEBUG    RETRIEVE SERVICES:
[Sat Mar 06 2021 14:25:02.187]  DEBUG    {"characteristics":[{"service":"180A","characteristic":"2A29","isNotifying":false,"properties":["Read"]},{"service":"180A","characteristic":"2A26","isNotifying":false,"properties":["Read"]},{"service":"180A","characteristic":"2A28","isNotifying":false,"properties":["Read"]}],"id":"204B2876-CEAC-62FC-9B8E-6088E73B78CD","advertising":{"serviceUUIDs":["1812"],"isConnectable":true,"localName":"Spionglas"},"name":"Spionglas","rssi":-60,"services":["180A","CCE20000-3F28-41CA-B995-FBBA51782E8F"]}
[Sat Mar 06 2021 14:25:03.207]  DEBUG    START NOTIFICATION:
[Sat Mar 06 2021 14:25:03.210]  DEBUG    Notification error ------  Could not find characteristic with UUID cce20001-3f28-41ca-b995-fbba51782e8f on service with UUID CCE20000-3F28-41CA-B995-FBBA51782E8F on peripheral with UUID 204B2876-CEAC-62FC-9B8E-6088E73B78CD

Alexispap, 2021-03-08:
Yes, i call retrieveServices shortly after connecting with a setTimeout.

djain1992, 2021-03-08:
IOS code has logs. Did you try running it from Xcode and look at the debug logs. I see several debug statements:

- (void)peripheral:(CBPeripheral *)peripheral didDiscoverServices:(NSError *)error {
    if (error) {
        NSLog(@"Error: %@", error);
        return;
    }
    NSLog(@"Services Discover");
    
    NSMutableSet *servicesForPeriperal = [NSMutableSet new];
    [servicesForPeriperal addObjectsFromArray:peripheral.services];
    [retrieveServicesLatches setObject:servicesForPeriperal forKey:[peripheral uuidAsString]];
    for (CBService *service in peripheral.services) {
        NSLog(@"Service %@ %@", service.UUID, service.description);
        [peripheral discoverCharacteristics:nil forService:service]; // discover all is slow
    }
}

Alexispap, 2021-03-08:
In XCode it only says: "Terminated due to signal 9"

Alexispap, 2021-03-08:
2021-03-08 18:35:30.418595+0100 frexitcc2[963:1986151] BleManager created
2021-03-08 18:35:30.712633+0100 frexitcc2[963:1986364] TIC Read Status [2:0x0]: 1:57
2021-03-08 18:35:30.716225+0100 frexitcc2[963:1986364] TIC Read Status [2:0x0]: 1:57
2021-03-08 18:35:32.059917+0100 frexitcc2[963:1986409] [] nw_connection_get_connected_socket 7 Connection has no connected handler
2021-03-08 18:35:32.061210+0100 frexitcc2[963:1986409] TCP Conn 0x10f3343b0 Failed : error 0:61 [61]

...

{ id: '204B2876-CEAC-62FC-9B8E-6088E73B78CD',
    rssi: -65,
    advertising: 
     { serviceUUIDs: [ '1812' ],
       isConnectable: true,
       localName: 'Spionglas' },
    name: 'Spionglas' },
  { id: '0FF35231-51E6-F125-EFAF-8B299E7D3918',
    rssi: -85,
    advertising: 
     { isConnectable: false,
       manufacturerData: 
        { CDVType: 'ArrayBuffer',
          data: 'BgABCSACWrA0Mky4QgGP0H3QQrLH4E5Qnx5f/Fk=',
          bytes: 
           [ 6,
             0,
             1,
             9,
             32,
             2,
             90,
             176,
             52,
             50,
             76,
             184,
             66,
             1,
             143,
             208,
             125,
             [TOO BIG formatValueCalls 201 exceeded limit of 200],
             [TOO BIG formatValueCalls 202 exceeded limit of 200],
             [TOO BIG formatValueCalls 203 exceeded limit of 200],
             [TOO BIG formatValueCalls 204 exceeded limit of 200],
             [TOO BIG formatValueCalls 205 exceeded limit of 200],
             [TOO BIG formatValueCalls 206 exceeded limit of 200],
             [TOO BIG formatValueCalls 207 exceeded limit of 200],
             [TOO BIG formatValueCalls 208 exceeded limit of 200],
             [TOO BIG formatValueCalls 209 exceeded limit of 200],
             [TOO BIG formatValueCalls 210 exceeded limit of 200],
             [TOO BIG formatValueCalls 211 exceeded limit of 200],
             [TOO BIG formatValueCalls 212 exceeded limit of 200] ] } },
    name: [TOO BIG formatValueCalls 213 exceeded limit of 200] },
  [TOO BIG formatValueCalls 214 exceeded limit of 200],
  [TOO BIG formatValueCalls 215 exceeded limit of 200],
  [TOO BIG formatValueCalls 216 exceeded limit of 200],
  [TOO BIG formatValueCalls 217 exceeded limit of 200],
  [TOO BIG formatValueCalls 218 exceeded limit of 200] ]
2021-03-08 18:37:39.024065+0100 frexitcc2[963:1986423] [javascript] Couch-Hintergrund ist keins der Items
2021-03-08 18:37:39.024708+0100 frexitcc2[963:1986423] [javascript] Hue Lamp ist keins der Items
2021-03-08 18:37:39.025697+0100 frexitcc2[963:1986423] [javascript] NO NAME ist keins der Items
2021-03-08 18:37:39.027004+0100 frexitcc2[963:1986423] [javascript] NO NAME ist keins der Items
2021-03-08 18:37:39.030761+0100 frexitcc2[963:1986423] [javascript] NO NAME ist keins der Items
2021-03-08 18:37:39.031436+0100 frexitcc2[963:1986423] [javascript] NO NAME ist keins der Items
2021-03-08 18:37:39.034823+0100 frexitcc2[963:1986423] [javascript] Verbinde mit peripheral{"id":"204B2876-CEAC-62FC-9B8E-6088E73B78CD","rssi":-65,"advertising":{"serviceUUIDs":["1812"],"isConnectable":true,"localName":"Spionglas"},"name":"Spionglas"}
2021-03-08 18:37:39.035352+0100 frexitcc2[963:1986423] [javascript] Connect
2021-03-08 18:37:39.044866+0100 frexitcc2[963:1986423] [javascript] NO NAME ist keins der Items
2021-03-08 18:37:39.045294+0100 frexitcc2[963:1986423] [javascript] NO NAME ist keins der Items
2021-03-08 18:37:39.052116+0100 frexitcc2[963:1986423] [javascript] NO NAME ist keins der Items
2021-03-08 18:37:39.052956+0100 frexitcc2[963:1986423] [javascript] Hue Lamp ist keins der Items
[... 4 lines cut ...]
2021-03-08 18:37:39.801247+0100 frexitcc2[963:1986151] Peripheral Connected: 204B2876-CEAC-62FC-9B8E-6088E73B78CD
2021-03-08 18:37:39.828346+0100 frexitcc2[963:1986423] [javascript] Connected to 204B2876-CEAC-62FC-9B8E-6088E73B78CD
2021-03-08 18:37:40.753655+0100 frexitcc2[963:1987137] retrieveServices (null)
2021-03-08 18:37:40.759906+0100 frexitcc2[963:1986151] Services Discover
2021-03-08 18:37:40.762128+0100 frexitcc2[963:1986151] Service Device Information <CBService: 0x10f370e10, isPrimary = YES, UUID = Device Information>
2021-03-08 18:37:40.763373+0100 frexitcc2[963:1986151] Service CCE20000-3F28-41CA-B995-FBBA51782E8F <CBService: 0x11270c100, isPrimary = YES, UUID = CCE20000-3F28-41CA-B995-FBBA51782E8F>
2021-03-08 18:37:40.785515+0100 frexitcc2[963:1986151] Characteristics For Service Discover
2021-03-08 18:37:40.786029+0100 frexitcc2[963:1986151] Characteristics For Service Discover
2021-03-08 18:37:40.797259+0100 frexitcc2[963:1986423] [javascript] RETRIEVE SERVICES:
2021-03-08 18:37:40.798316+0100 frexitcc2[963:1986423] [javascript] {"characteristics":[{"service":"180A","characteristic":"2A29","isNotifying":false,"properties":["Read"]},{"service":"180A","characteristic":"2A26","isNotifying":false,"properties":["Read"]},{"service":"180A","characteristic":"2A28","isNotifying":false,"properties":["Read"]}],"id":"204B2876-CEAC-62FC-9B8E-6088E73B78CD","advertising":{"serviceUUIDs":["1812"],"isConnectable":true,"localName":"Spionglas"},"name":"Spionglas","rssi":-65,"services":["180A","CCE20000-3F28-41CA-B995-FBBA51782E8F"]}
2021-03-08 18:37:40.817829+0100 frexitcc2[963:1987137] writeWithoutResponse
2021-03-08 18:37:40.818431+0100 frexitcc2[963:1987137] Looking for CCE20001-3F28-41CA-B995-FBBA51782E8F with properties 4
[... cut ...] 9B8E-6088E73B78CD'
2021-03-08 18:37:42.843840+0100 frexitcc2[963:1986423] [javascript] Possible Unhandled Promise Rejection (id: 0):
"Could not find characteristic with UUID cce20001-3f28-41ca-b995-fbba51782e8f on service with UUID CCE20000-3F28-41CA-B995-FBBA51782E8F on peripheral with UUID 204B2876-CEAC-62FC-9B8E-6088E73B78CD"
2021-03-08 18:39:31.480502+0100 frexitcc2[963:1987717] TIC Read Status [1:0x0]: 1:57
2021-03-08 18:39:31.480696+0100 frexitcc2[963:1987717] TIC Read Status [1:0x0]: 1:57

marcosinigaglia, 2021-03-08:
Try add a delay also after the retrieveServices and the startNotification.

Alexispap, 2021-03-08:
It is still giving me the same messages :/

Alexispap, 2021-03-09:
Yes, it is custom. Do we need to use specific UUIDs or something like that so that Apple can observe it? It is very strange, because in any case it should not crash... Thanks for your help so far, that is very kind!

Alexispap, 2021-03-09:
I will try that as soon as i have an Iphone where i can install lightblue (at the moment i borrowed one and cannot install things - my own will come in the next days)

But i found this log in Xcode:

2021-03-10 00:09:18.115477+0100 frexitcc2[1031:2093847] BleManager initialized
2021-03-10 00:09:18.116456+0100 frexitcc2[1031:2093847] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x10b5c04d0> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
2021-03-10 00:09:18.123341+0100 frexitcc2[1031:2094367] [CoreBluetooth] XPC connection invalid

What does that mean?

djain1992, 2021-03-10:
This is ok. You can ignore this warning.

Alexispap, 2021-03-11:
[image]
[image]

Alexispap, 2021-03-11:
Does this mean, that the characteristic is somehow not correct in repect to iOS principles?

djain1992, 2021-03-11:
Yes. Not an issue with BLE manager package.
Which BLE chip do you use? How do you define your characteristics?

Alexispap, 2021-03-11:
We use an Seeeduino Wio Lite MG126 microcontroller and Arduino ide for implementing. Since i did not write the code, i have to understand it and find the correct lines before i can answer your question regarding the definition of the characteristics. Here is a part of the code, does that already help answer it?

//
// STEP 0: Character declare
//
const BLE_CHAR AttCharList[] = {
  // =====  GATT =====
  {TYPE_CHAR, 0x0003, {ATT_CHAR_PROP_RD, 0x04, 0, 0x00, 0x2a}, UUID16_FORMAT}, //name
  //05-06 reserved

  // ===== Device Information Service Characteristics =====
  {TYPE_CHAR, 0x0008, {ATT_CHAR_PROP_RD, 0x09, 0, 0x29, 0x2a}, UUID16_FORMAT}, //manufacture
  {TYPE_CHAR, 0x000a, {ATT_CHAR_PROP_RD, 0x0b, 0, 0x26, 0x2a}, UUID16_FORMAT}, //firmware version
  {TYPE_CHAR, 0x000e, {ATT_CHAR_PROP_RD, 0x0f, 0, 0x28, 0x2a}, UUID16_FORMAT}, //sw version

  // ===== Item Service Characteristics =====
  {TYPE_CHAR, 0x20, {ATT_CHAR_PROP_RD | ATT_CHAR_PROP_W | ATT_CHAR_PROP_W_NORSP | ATT_CHAR_PROP_NTF, 0x21, 0, 0}, 1}, // Characteristic
  {TYPE_INFO, 0x22},                                                                              // Characteristic User Description
  {TYPE_CFG, 0x23, {ATT_CHAR_PROP_RD|ATT_CHAR_PROP_W}},                                           // Client Characteristic Configuration
  {TYPE_PRESENTATION_FORMAT, 0x24},                                                               // Characteristic Pre-sentation Format
};

// List of 128-bit UUIDs
const BLE_UUID128 AttUuid128List[] = {
  {0x8f, 0x2e, 0x78, 0x51, 0xba, 0xfb, 0x95, 0xb9, 0xca, 0x41, 0x28, 0x3f, 0x00, 0x00, 0xe2, 0xcc},
  {0x8f, 0x2e, 0x78, 0x51, 0xba, 0xfb, 0x95, 0xb9, 0xca, 0x41, 0x28, 0x3f, 0x01, 0x00, 0xe2, 0xcc},
};

djain1992, 2021-03-12:
Only thing I can think of is the connection handle are not continuous. Can
you try starting the connection handle from 0x10 instead of 0x20.

If this does not work try starting from 0x0f.

Alexispap, 2021-03-12:
Thank you, i will try it :)

Alexispap, 2021-03-14:
I cannot fix this :/ Do you have any other idea?

Alexispap, 2021-03-14:
The full file is now:

/// Characteristic Properties Bit
#define ATT_CHAR_PROP_RD                            0x02
#define ATT_CHAR_PROP_W_NORSP                       0x04
#define ATT_CHAR_PROP_W                             0x08
#define ATT_CHAR_PROP_NTF                           0x10
#define ATT_CHAR_PROP_IND                           0x20
#define GATT_PRIMARY_SERVICE_UUID                   0x2800

#define TYPE_PRESENTATION_FORMAT  0x2904
#define TYPE_CHAR                 0x2803
#define TYPE_CFG                  0x2902
#define TYPE_INFO                 0x2901
#define TYPE_xRpRef               0x2907
#define TYPE_RpRef                0x2908
#define TYPE_INC                  0x2802
#define UUID16_FORMAT             0xff

// ===== Device Information Service =====
#define SOFTWARE_INFO "v1.2"
#define MANU_INFO     "FREXIT"
// ======================================

// ===== GAP =====
char DeviceInfo[9] =  "Spionglas";  /*max len is 24 bytes*/
// ===============

// ===== Item Service =====
char itemCharacteristicUserDescription[18] = "Show item solution";
byte cccd[2] = { 0, 0 };
// Format:1, Exponent:1, Unit:2, NameSpace:1, Description:2
byte itemCharacteristicPresentationFormat[7] = {0x01, 0x00, 0x00, 0x27};

uint16_t cur_notifyhandle = 0x21;

uint8_t* getDeviceInfoData(uint8_t* len)
{
  *len = sizeof(DeviceInfo);
  return (uint8_t*)DeviceInfo;
}

void updateDeviceInfoData(uint8_t* name, uint8_t len)
{
  memcpy(DeviceInfo, name, len);
  ble_set_name(name, len);
}

/**********************************************************************************
                 *****DataBase****

  01 - 06  GAP (Primary service) 0x1800
  03:04  name
  07 - 0f  Device Info (Primary service) 0x180a
  0a:0b  firmware version
  0e:0f  software version
  10 - 19  LED service (Primary service) 6E400001-B5A3-F393-E0A9-E50E24DCCA9E
  11:12  6E400003-B5A3-F393-E0A9-E50E24DCCA9E(0x04)  RxNotify
  13     cfg
  14:15  6E400002-B5A3-F393-E0A9-E50E24DCCA9E(0x0C)  Tx
  16     cfg
  17:18  6E400004-B5A3-F393-E0A9-E50E24DCCA9E(0x0A)  BaudRate
  19     0x2901  info
************************************************************************************/

typedef struct ble_character16 {
  uint16_t type16;          //type2
  uint16_t handle_rec;      //handle
  uint8_t  characterInfo[5];//property1 - handle2 - uuid2
  uint8_t  uuid128_idx;     //0xff means uuid16,other is idx of uuid128
[... 174 lines cut ...]

char GetConnectedStatus(void)
{
  return gConnectedFlag;
}

void ConnectStausUpdate(unsigned char IsConnectedFlag) //porting api
{
  LED_ONOFF(!IsConnectedFlag);

  if (IsConnectedFlag != gConnectedFlag)
  {
    gConnectedFlag = IsConnectedFlag;
  }
}

void UsrProcCallback(void) //porting api
{
  return;
}

unsigned char aes_encrypt_HW(unsigned char *_data, unsigned char *_key)
{
  return 0;
}

Alexispap, 2021-03-19:
Some progress: the characteristic can be found now after changing the handles many times :) But now the peripheral disconnects shortly after writing the value on both platforms. LightBlue gives the message "Disconnected. Data is stale". May i ask you again for ideas?

closed 2023-12-06 by marcosinigaglia