innoveit/react-native-ble-manager #719: Error: EISDIR: illegal operation on a directory, read

withgogo, 2021-03-17:
### Before open an issue
- Check the closed issues, your question maybe is not new.
- We can't debug your physical devices.
- If the library is not working 99% you did something wrong, in the code or in the phone permissions.

### Version

Tell us which versions you are using:

- react-native-ble-manager v7.4.2
- react-native v0.64.0
- iOS/Android v14.4
- MacOS 11.2.1
### Expected behaviour

Run the example

### Actual behaviour

getting error

### Steps to reproduce

1. download example as the standalone project.
2. npm install && ./ios && pod install
3. react-native run-ios --simulator="iPhone 11" 
4. getting error

I also tried "chmod -R 777" to example folder

### Stack trace and console log

Error: EISDIR: illegal operation on a directory, read
    at Object.readSync (node:fs:617:3)
    at tryReadSync (node:fs:382:20)
    at Object.readFileSync (node:fs:419:19)
    at UnableToResolveError.buildCodeFrameMessage (/Users/sam/Dev/dev/RnDemo/node_modules/metro/src/node-haste/DependencyGraph/ModuleResolution.js:347:17)
    at new UnableToResolveError (/Users/sam/Dev/dev/RnDemo/node_modules/metro/src/node-haste/DependencyGraph/ModuleResolution.js:333:35)
    at ModuleResolver.resolveDependency (/Users/sam/Dev/dev/RnDemo/node_modules/metro/src/node-haste/DependencyGraph/ModuleResolution.js:211:15)
    at DependencyGraph.resolveDependency (/Users/sam/Dev/dev/RnDemo/node_modules/metro/src/node-haste/DependencyGraph.js:413:43)
    at /Users/sam/Dev/dev/RnDemo/node_modules/metro/src/lib/transformHelpers.js:317:42
    at /Users/sam/Dev/dev/RnDemo/node_modules/metro/src/Server.js:1471:14
    at Generator.next (<anonymous>)

```
```

withgogo, 2021-03-18:
sorry, forgo to link

closed 2021-03-18 by withgogo

hoanghuuf17, 2021-04-25:
> sorry, forgo to link

did you fix it?

jkoutavas, 2021-05-14:
Please give details on the fix.

kaixuannnn, 2021-06-10:
I fix it by deleting node module and "yarn install" all the package again.