innoveit/react-native-ble-manager #720: Infinite loop on read

danijelmartinek, 2021-03-20:
### Version

- react-native-ble-manager v7.4.2
- react-native v0.62.2
- Android v10

### Expected behaviour

Bluetooth module returns data once.

### Actual behaviour

Bluetooth module goes into infinite loop and returns same data over and over again

### Steps to reproduce

1. Connect to BT module
2. Setup service and characteristics for read and write, start notification
3. Run write or writeWithoutResponse functions

First, some info for context. I'm developing an application which connects to external bluetooth module which is connected to microcontroller. BT module accepts read commands in the form of one character (for example 'b' for reading battery status), and returns string which contains some data. I'm developing just an app and this other developer programs module and microcontroller.
Before he made some changes to the module (controller part, not BT) everything worked well, but after that changes when I write some char, instead of returning data once it keeps returning data infinitely, as if in some while loop. I asume that there is some problem with emptying buffer, but can't detect what. At first, we were thinking that problem was on the module, but he made some changes and problem is still here. Strange thing is when testing with "Serial Bluetooth Terminal" android app (https://github.com/kai-morich/SimpleBluetoothTerminal) everything works as it should, and when he tests on his computer everything works also as it should.
I have also tried using react-native-ble-plx lib but infinite loop still occurs.
Some note that may be helpful, one change that he made compared to the version before that worked is that module no longer sends entire string to an app but single characters of that string with delay between each another.

marcosinigaglia, 2021-03-22:
Hi, maybe is something wrong in how you handle the BleManagerDidUpdateValueForCharacteristic event.

danijelmartinek, 2021-03-22:
I found a problem. Forgot carriage return and line feed chars (CR + LF).

For everyone having the same problem just add _\r\n_ to end of write string.

e.g. .write(writeData + '\r\n')

closed 2021-03-26 by marcosinigaglia