innoveit/react-native-ble-manager #721: How to request MTU (Maximum Transmission Unit) for iOS using react-native-ble-manager

onn1k, 2021-03-24:
**Version**
react-native-ble-manager v7.4.1
react-native v0.64.0
iOS all versions

**Expected behaviour**
After writing a value in a specific characteristic, the app will get a response

**Actual behaviour**
the app will not get a response from characteristic on ios device

**Steps what I am doing:**
search the device
connect to the device
enable notifications
set MTU (Android only API 21+)
send value (1)

**Part of code:**
`const handleConnectPeripheral = (device) => {
  console.log("BleManagerConnectPeripheral:", device);

  BleManager.isPeripheralConnected(deviceId, [])
    .then((isConnected) => {
      if (isConnected) {
        console.log("Peripheral is connected!");

        setTimeout(() => {
          BleManager.retrieveServices(device.peripheral).then((data) => {
            console.log("retrieveServices: ", data.services);

            BleManager.startNotification(
              data.id,
              deviceInfo.service,
              deviceInfo.shortNtfChar
            )
              .then(() => {
                console.log('ShortNotification on');

                BleManager.startNotification(
                  data.id,
                  deviceInfo.service,
                  deviceInfo.longNotfChar
                )
                  .then(() => {
                    console.log('LongNotification on')

                    if (Platform.OS === 'android') {
                      BleManager.requestMTU(data.id, 400) //android only
                        .then((mtu) => {
                          console.log('MTU size changed to ' + mtu + ' bytes');
                        })
                        .catch((error) => {
                          console.error('maxbytes: ', error);
                        });
                    }
                    console.log('!!!You can start to send commands!!!');
                  })
                  .catch((error) => console.log('LongNotification', error));
              })
              .catch((error) => console.log('ShortNotification', error));

          }).catch((error) => console.log(error));
        }, 900);
      } else {
        console.log("Peripheral is NOT connected!");
      }
    });
};

const writeCmd = (cmd) => { //click button 'send command', cmd = [1];
    BleManager.write(deviceId, deviceInfo.service, deviceInfo.char, [cmd])
    .then(() => {
      console.log(`Command ${cmd} was sent successfully.`);
    })
    .catch((err) => {
      console.log(err);
    })
}

const handleUpdateValueForCharacteristic = ({ value }) => { //bind
  console.log("handleUpdateValueForCharacteristic");

  switch (value[0]) {
    case commands.readStatusCmd: //[1]
      console.log('response: ', JSON.stringify(value));
      ...

      break;
...
`

**Logs from xCode**
`2021-03-24 16:34:25.752114+0200 emeter_testing_environment[1109:287196] [javascript] Connecting to device...
2021-03-24 16:34:25.799624+0200 emeter_testing_environment[1109:287189] Connect
2021-03-24 16:34:25.799862+0200 emeter_testing_environment[1109:287189] Connecting to peripheral with UUID : 2D3F171A-15BE-0C3C-6729-EA55779F5555
2021-03-24 16:34:30.885985+0200 emeter_testing_environment[1109:286991] Peripheral Connected: 2D3F171A-15BE-0C3C-6729-EA55779F5555
2021-03-24 16:34:30.893482+0200 emeter_testing_environment[1109:287196] [javascript] Connected
2021-03-24 16:34:30.896836+0200 emeter_testing_environment[1109:287196] [javascript] 'BleManagerConnectPeripheral:', { peripheral: '2D3F171A-15BE-0C3C-6729-EA55779F5555' }
2021-03-24 16:34:30.899936+0200 emeter_testing_environment[1109:287189] Get connected peripherals
2021-03-24 16:34:30.903288+0200 emeter_testing_environment[1109:287196] [javascript] Peripheral is connected!
2021-03-24 16:34:31.861726+0200 emeter_testing_environment[1109:287189] retrieveServices (null)
2021-03-24 16:34:31.866628+0200 emeter_testing_environment[1109:286991] Services Discover
2021-03-24 16:34:31.867302+0200 emeter_testing_environment[1109:286991] Service 12345687-0000-1000-8000-00805F123456 <CBService: 0x28232b110, isPrimary = YES, UUID = 12345687-0000-1000-8000-00805F123456>
2021-03-24 16:34:31.934122+0200 emeter_testing_environment[1109:286991] Characteristics For Service Discover
2021-03-24 16:34:31.938079+0200 emeter_testing_environment[1109:287196] [javascript] 'retrieveServices: ', [ '12345687-0000-1000-8000-00805F9B34FB' ]
2021-03-24 16:34:31.940274+0200 emeter_testing_environment[1109:287189] startNotification
2021-03-24 16:34:31.941162+0200 emeter_testing_environment[1109:287189] Looking for FC25 with properties 16
2021-03-24 16:34:31.942065+0200 emeter_testing_environment[1109:287189] Found FC25
2021-03-24 16:34:32.050390+0200 emeter_testing_environment[1109:286991] Notification began on FC25
2021-03-24 16:34:32.052233+0200 emeter_testing_environment[1109:287196] [javascript] ShortNotification on
2021-03-24 16:34:32.085684+0200 emeter_testing_environment[1109:287365] startNotification
2021-03-24 16:34:32.086259+0200 emeter_testing_environment[1109:287365] Looking for AEF8 with properties 16
2021-03-24 16:34:32.086658+0200 emeter_testing_environment[1109:287365] Found AEF8
2021-03-24 16:34:32.199429+0200 emeter_testing_environment[1109:286991] Notification began on AEF8
2021-03-24 16:34:32.200781+0200 emeter_testing_environment[1109:287196] [javascript] LongNotification on
2021-03-24 16:34:32.230039+0200 emeter_testing_environment[1109:287196] [javascript] !!!You can start to send commands!!!
2021-03-24 16:34:32.280096+0200 emeter_testing_environment[1109:287196] [javascript] 'maxbytes: ', 'Not supported'
2021-03-24 16:34:35.934612+0200 emeter_testing_environment[1109:287189] Write
2021-03-24 16:34:35.935617+0200 emeter_testing_environment[1109:287189] Looking for 8AA2 with properties 8
2021-03-24 16:34:35.936552+0200 emeter_testing_environment[1109:287189] Found 8AA2
2021-03-24 16:34:35.937259+0200 emeter_testing_environment[1109:287189] [native] Message to write(1): (
    1
)
2021-03-24 16:34:35.937563+0200 emeter_testing_environment[1109:287189] [native] Message to write(1): 01
2021-03-24 16:34:35.979833+0200 emeter_testing_environment[1109:286991] didWrite
2021-03-24 16:34:35.982383+0200 emeter_testing_environment[1109:287196] [javascript] Command 1 was sent successfully.`

**_...But code works on Android devices._** 

I suggest it happened because the module has not requested MTU for iOS, right? How I can fix it?

Any ideas?

Thanks.

onn1k, 2021-03-25:
I found the post: [(https://qiita.com/hituziando/items/50c7ba7fb9f8917e1ebf)]
But I have not understood what I have to change...

marcosinigaglia, 2021-03-26:
You can't change the MTU on iOS

closed 2021-03-26 by marcosinigaglia