innoveit/react-native-ble-manager #722: Location access in background

siddaiahm, 2021-04-03:
We have built an app for ble functions like (scan, connect, write and read data from ble devices) using this module. This app runs in both foreground and background. We are accessing location only in foreground but all ble functions runs in both foreground and background. Till last app release everything was working fine, regarding the current [Google Play Policy ](https://support.google.com/googleplay/android-developer/answer/9799150) about location access permissions  our app got rejected by the Play store saying (reason: We detected that your app Bundle contains at least one feature that requests access to location in the background ). 
I see that this module requires 

> ( Android API >= 23 require the ACCESS_COARSE_LOCATION permission to scan for peripherals. React Native >= 0.33 natively support PermissionsAndroid like in the example.
> Android API >= 29 require the ACCESS_FINE_LOCATION permission to scan for peripherals. React-Native 0.63.X started targeting Android API 29. )

 I also found this from android developer documentation 

> Because discoverable devices might reveal information about the user's location, the device discovery process requires location access. If your app is being used on a device that runs Android 8.0 (API level 26) or higher, consider using the Companion Device Manager API, instead. This API performs device discovery on your app's behalf, so that your app doesn't need to request location permissions.

If you don't use the [ Companion Device Manager API ](https://developer.android.com/guide/topics/connectivity/bluetooth-le), declare the ACCESS_FINE_LOCATION permission.

how can we continue with this !?
Thank you.

marcosinigaglia, 2021-04-16:
You have to declare the ACCESS_FINE_LOCATION in your manifest.

siddaiahm, 2021-04-16:
we have declared both ACCESS_FINE_LOCATION and ACCESS_COARSE_LOCATION.

marcosinigaglia, 2021-04-16:
So I don't understand the problem with the Play Store

franzejr, 2021-05-02:
@siddaiahm did you get it approved on the Play Store?

siddaiahm, 2021-05-04:
Not yet, but we have submitted a request for background location access.

closed 2023-12-04 by marcosinigaglia