innoveit/react-native-ble-manager #726: manufacturerData on Android returning incorrect bytes

gururanganarasimhan, 2021-04-12:
### Before open an issue
- Check the closed issues, your question maybe is not new
- We can't debug your physical devices
- If the library is not working 99% you did something wrong, in the code or in the phone permissions

### Version

Tell us which versions you are using:

- react-native-ble-manager v7.4.1
- react-native v4.13
- iOS/Android - Android 8

### Expected behaviour

In BleManagerDiscoverPeripheral callback, the field advertising.manufacturerData.bytes is expected to return only the Manufacturer data bytes, to be consistent with iOS

### Actual behaviour
It returns the raw bytes for the entire ScanRecord

### Steps to reproduce

1.  Setup a BLE device to advertise some manufacturer-specific data
2. Scan and find the device on iOS and Android
3. Compare the advertising.manufacturerData.bytes data in the BleManagerDiscoverPeripheral callback

### Stack trace and console log

The issue seems to be here 
https://github.com/innoveit/react-native-ble-manager/blob/e7d83a503fd351f7a84275748391bffd05e11a55/android/src/main/java/it/innove/LollipopPeripheral.java#L40

The above line should use the bytes returned by getManufacturerSpecificData() method of ScanRecord, instead of getBytes() from line https://github.com/innoveit/react-native-ble-manager/blob/e7d83a503fd351f7a84275748391bffd05e11a55/android/src/main/java/it/innove/LollipopPeripheral.java#L87

gururanganarasimhan, 2021-04-12:
https://developer.android.com/reference/android/bluetooth/le/ScanRecord#getManufacturerSpecificData()

enchorb, 2022-12-07:
These two are probably related https://github.com/innoveit/react-native-ble-manager/issues/621

@marcosinigaglia can this be fixed.

I tried to fix it with the the patch below, but then when building got `incompatible types: SparseArray<byte[]> cannot be converted to byte[]`. I do not know enough about native Android development to fix this.

```
diff --git a/node_modules/react-native-ble-manager/android/src/main/java/it/innove/LollipopPeripheral.java b/node_modules/react-native-ble-manager/android/src/main/java/it/innove/LollipopPeripheral.java
index 4113087..bcfa045 100644
--- a/node_modules/react-native-ble-manager/android/src/main/java/it/innove/LollipopPeripheral.java
+++ b/node_modules/react-native-ble-manager/android/src/main/java/it/innove/LollipopPeripheral.java
@@ -86,7 +86,7 @@ public class LollipopPeripheral extends Peripheral {
 
 	public void updateData(ScanResult result) {
 		advertisingData = result.getScanRecord();
-		advertisingDataBytes = advertisingData.getBytes();
+		advertisingDataBytes = advertisingData.getManufacturerSpecificData();
 	}
```

ferkawi, 2023-05-26:
@enchorb I think is because in the `android/src/main/java/it/innove/Peripheral.java` is the wrong type:
 
```
public void updateData(byte[] data) {
    advertisingDataBytes = data;
 }
```

Try to change it to `SparseArray<byte[]>` and if it's working let's do a PR! I need this change too, to make the app consistent and get away from workarounds :)