innoveit/react-native-ble-manager #734: Feature Request: integration of queue management system

WookieFPV, 2021-05-19:
BLE operation can't happen in parallel.
Instead of implementing a queue system in the app, it would be nice if the ble-manager could manage the queuing.

I have to create a queue system either way.
I think It would be nice to directly contribute this to the ble-manager.

@marcosinigaglia Do you think this does make sense?
Or should I just create the queue system in the App and not in this Library?

The proposed change wouldn't change the external API of the ble-manager.

For example, If a write is already in progress and another write is called It would just be placed in the queue and the promise would be resolve once the operation is done.

marcosinigaglia, 2021-05-19:
Hi @WookieFPV, there is already a PR (https://github.com/innoveit/react-native-ble-manager/pull/660) for Android but it's not complete. It would certainly be an interesting feature. If you want to contribute to the library you are welcome.

WookieFPV, 2021-05-19:
thanks for the fast response,
I am going to implement this feature on the JS side.

closed 2021-11-26 by marcosinigaglia