innoveit/react-native-ble-manager #735: retrieveServices doesn't respond for a long time

Tanisi001, 2021-05-24:
Hi,
We tried adding a custom timeout of 15 seconds but still faced this issues.
We can see that in the first attempt, BLEManager.retreiveServices took around 14 seconds to return and our timeout is 15 seconds which led to timeout.
In the subsequent attempt, BLEManager.retreiveServices did not return any response till 15 seconds.

Used:
- react-native-ble-manager v6.6.2
- react-native v0.59.8
- DeviceModel: iPhone12,1, version: 14.6

Is there any solution for this?

_Originally posted by @Tanisi001 in https://github.com/innoveit/react-native-ble-manager/issues/185#issuecomment-846731038_

Spoki4, 2021-09-02:
Something unknown reason, `retrieveServices` never resolved or rejected.

I fixed that by remove await from `retrieveServices` and `setTimeout` after starting `retrieveServices`.
Example of work code:
```
await BleManager.connect(peripheral)
BleManager.retrieveServices(peripheral) // no await specially

setTimeout(async () => {  
  await BleManager.startNotification(....);
  await BleManager.read(....);
  ...more...
}, 1000)
```

ivokutlesa, 2021-09-08:
> Something unknown reason, `retrieveServices` never resolved or rejected.
> 
> I fixed that by remove await from `retrieveServices` and `setTimeout` after starting `retrieveServices`.
> Example of work code:
> 
> ```ts
> await BleManager.connect(peripheral)
> BleManager.retrieveServices(peripheral) // no await specially
> 
> setTimeout(async () => {  
>   await BleManager.startNotification(....);
>   await BleManager.read(....);
>   ...more...
> }, 1000)
> ```

@Spoki4 this solution is not working because if retriveServises is never resolved or rejected, startNotification will not work. I get "Possible Unhandled Promise Rejection (id: 0): Characteristic not found" warning and device is not connected.

closed 2023-12-20 by marcosinigaglia