innoveit/react-native-ble-manager #736: requestMTU success, but still receieved 20 characters

15638228357, 2021-05-26:
BluetoothManager.connect(this.peripheralId)
      .then(() => {
        console.log('Connected');
        BluetoothManager.requestMTU(200);
        setTimeout(() => {
          this.getNotic();
        }, 500);
      })
      .catch(error => {
        console.log('Connected error:', error);
      });

jaimebg, 2022-01-27:
requestMTU doesn't seem to be doing anything version 7.6.1, also noticed packages are still 20 bytes.
On 8.0.1 the Promise never returns... it gets stuck in that line

AlexManSPB, 2022-08-04:
try setting the MTU value to 185 or less. It worked for me

AlexManSPB, 2023-07-28:
> I am still facing the same issue. I'm trying to set MTU to 512 it shows ok in logs but actually it still sending 20 bytes chunks. Please help if someone resolved this issue

Hi! Experimentally, I found that in some Android systems, the MTU of more than 185 does not work. Try to make the value no more than 185.
My code example:
...
const TimePauseMTU = 2000; // необходима задержка после соединения (перед установкой MTU)

await BleManager.connect(peripheral.id);

await new Promise((resolve, _reject) =>
        setTimeout(resolve, TimePauseMTU),
      );

if (Platform.OS === 'android') {
        await BleManager.requestMTU(peripheral.id, 185)
	}

await BleManager.retrieveServices(peripheral.id);
...

HamzaQayyumMST, 2023-07-31:
@AlexManSPB I tried but it didn't worked 
is there another solution. 
should i use react native ble -plex?

AlexManSPB, 2023-08-18:
> @AlexManSPB I tried but it didn't worked is there another solution. should i use react native ble -plex?

I think react native ble-plx won't solve your issue. A possible solution is to accept data in several smaller packets

closed 2023-12-04 by marcosinigaglia