innoveit/react-native-ble-manager #750: Scan returns 0 results even with the sample

sssembler, 2021-06-22:
Samsung S21 Ultra, trying the sample app as is. nothing.

viren-vii, 2021-06-24:
for me, scan returns results as undefined! How are you getting even 0?

Anggasayogo, 2021-07-02:
yes i ran into this after using the sample

marcosinigaglia, 2021-07-07:
Hi, if is not working it's a installation/device/hw problem.

closed 2021-07-07 by marcosinigaglia

sssembler, 2021-07-07:
not really a valid response, it happens with your default test sample with multiple users

marcosinigaglia, 2021-07-07:
@sssembler the library use the OS api so if you don't have result:

- the app don't have the right permissions
- there is an issue in the phone
- there is an issue in OS
- there is an issue on the BLE device

There is no way that anyone can reproduce your problem without the combination of phone/os/ble device.

sssembler, 2021-07-07:
ble plx worked first time so what can I say.

marcosinigaglia, 2021-07-09:
@sssembler i'm happy the plx library is working and I can't debug your device and reproduce your problem.