innoveit/react-native-ble-manager #754: StopNotification problem

sale7268, 2021-06-28:
### Before open an issue
- Check the closed issues, your question maybe is not new.
- We can't debug your physical devices.
- If the library is not working 99% you did something wrong, in the code or in the phone permissions.

### Version

Tell us which versions you are using:

react-native-ble-manager v7.5.0
react-native v0.60
iOS/Android v.14

### Expected behaviour

After reconnection i want it to read value once and before starting notification i want it to stop notification.

### Actual behaviour

When app connects for the first time it gives correct values like below: 
`CV (sdjkasgdsajmbfjsahfsa)`
But when i go back to other screen and navigate back to BLE connection screen it automatically disconnects me from the device and when i connect again it gives me abnormal values like this:
`CV (sdjkasgdsajmbfjsahfsa)CV (sdjkasgdsajmbfjsahfsa)`

Same value received twice. Can't figure out where to put stopNotifications and keep BLE connected when screen unmounts. 

### Steps to reproduce

1. Too many steps to reproduce, not possible to write all, but simple would be:
2. Create 2 screens using stack navigation. 
3. One with BLE connection and everything and other one can be just empty screen
4. Navigate back to BLE screen after connection to check if it disconnects or not
5. Check if StopNotification will work before transitioning into BLE connection screen 2nd time.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

marcosinigaglia, 2021-07-05:
Hi, you have to put all the BLE stuff in a "global" component.

JClackett, 2021-07-09:
@marcosinigaglia What do you mean by all the BLE stuff? if you have multiple places in the app where you connect to a device, how would you handle this?

MaxHa-28, 2021-07-09:
> @marcosinigaglia What do you mean by all the BLE stuff? if you have multiple places in the app where you connect to a device, how would you handle this?

E.g. you can use redux with thunks or sagas to scan, connect or do read/write operations. Your ble related components only need to dispatch "actions" and the state of your app is saved in a global store.

JClackett, 2021-07-09:
Hmm is there any documentation regarding this, or somewhere in the source code that shows that this needs to happen? I have a working app right now that doesn't use context to pass BleManager around. It seems strange that it has to be set up this way. Wouldn't a singleton would work in that case i.e just import and export BleManager from a file and use that everywhere instead of importing from the package itself? Thanks for responding though, super curious if this is the recommended approach

MaxHa-28, 2021-07-09:
Well I am not sure if there is a recommended way of handling ble. Most of the time it really depends on the app your want to build.

But in the end if you need state management for your app aka you want to save and access data in a global manner, i would recommend using a state management library and integrate the BleManager!

MaxHa-28, 2021-07-09:
Well, today i was tinkering with this lib and redux+saga, but was caught up by bluez and dbus.. bluetooth is literal hell :D

But I uploaded my test-Repo from my days work, it is quite chaotic but I hope you can see how i implemented sagas (I am no expert - But it works pretty good). If you find a good solution for your case, please keep me posted! I am also interested in the recommened approach.

[Repo](https://github.com/MaxHa-28/dev_rn_jordan)

Have a nice weekend!

closed 2023-12-04 by marcosinigaglia