innoveit/react-native-ble-manager #755: startNotification gives error on iOS device

Jwlee134, 2021-07-08:
### Before open an issue
- Check the closed issues, your question maybe is not new.
- We can't debug your physical devices.
- If the library is not working 99% you did something wrong, in the code or in the phone permissions.

### Version

Tell us which versions you are using:

- react-native-ble-manager v7.5.0
- react-native v0.64.1
- iOS v14.6

### Expected behaviour

startNotification should be working.

### Actual behaviour

when startNotification invokes it gives error: **Peripheral did disconnect: F60E08DE-5A03-0C18-DDF3-23D93AB1D0F2**

### Steps to reproduce

1. Connect to a peripheral
2. Retrieve services
3. Start notification

### Stack trace and console log

This is a part of code:
```
BleManager.connect(peripheral.id)
          .then(() => {
            console.log('Connected to ' + peripheral.id); // Success

            BleManager.retrieveServices(peripheral.id).then(peripheralInfo => {
              console.log(peripheralInfo); // Success

              BleManager.startNotification(
                peripheral.id,
                Service.UUID,
                Service.CharacteristicA,
              )
                .then(() => {
                  console.log('Started notification on ' + peripheral.id); // Fail
                })
                .catch(error => {
                  console.log('Notification error', error);
                });
            });
          })
```
I didn't forget to use the start method before anything.
Also this code works perfectly on my Android device, I have no idea what's going on my iOS device.
Thank you for your help.

vish188, 2021-07-08:
Can u share ur console. Problem may be with ur Service.UUID and Service.CharacteristicA, once check it by printing the data in console.

Jwlee134, 2021-07-08:
[image]

@vish188 Hi this is my console, why is the error not showing on android device if there's a problem with my UUIDs?

vish188, 2021-07-08:
Check your Hardware device because in console ("isNotifying : false) it means notification is not enabled in ur hardware.Try to reset ur (isNotifying : true) in Hardware device.

Jwlee134, 2021-07-23:
@vish188 Sorry for late, I enabled notification permission but it didn't work, do i need to add Push Notification capability in my xcode signing & capabilities section also?

josholdham, 2021-07-30:
I am seeing this issue, and I do notice that "isNotifying" is false as noted by @vish188. I'm not sure how to "reset (isNotifying : true) in Hardware device" though. Does anyone have any pointers?

I'm connecting from an ios tablet to a printer: it works with most printers but there is the odd one of the same make/model that it fails for.

Jwlee134, 2021-08-20:
Solved by changing gatt characteristic property problem.

closed 2021-08-20 by Jwlee134

Koxx3, 2022-05-29:
hello,
I have the same issue with "isNotifying" : false on ios.
it works fine on android... but impossible to cathc notifications on ios.
any idea why ?