innoveit/react-native-ble-manager #757: Autoconnect (re-connect)

ivokutlesa, 2021-07-12:
### Version

- react-native-ble-manager v7.6.0
- react-native v0.64.2
- Android v.10

### Help
I basically need this https://github.com/innoveit/react-native-ble-manager/issues/246 and the link in the comment [https://stackoverflow.com/questions/25254932/can-you-pair-a-bluetooth-le-device-in-an-ios-app](https://stackoverflow.com/questions/25254932/can-you-pair-a-bluetooth-le-device-in-an-ios-app) is not helping me and I cannot find solution.

To simplify: I would like to know is it possible to re-connect a device every time device is in the range, using react-native-ble-manager? And if yes, how? 
Lets assume that app is always on, so it doesn't have to save the device permanently (after restarting the app).

marcosinigaglia, 2021-08-06:
Hi, if the app is running you can continuously scan and search for the device.

closed 2022-08-12 by marcosinigaglia

shehanhasanga, 2022-09-07:
that means this library does mot support for auto connect like android sdk. right? @marcosinigaglia